jdesbonnet/RCWL-0516 #1: measured power usage

ohsix, 2017-03-28:
it's 2.95-3.05 mA at 5v ;]

hurvajs, 2017-06-11:
My measurements (2 units, nothing connected just power):
3,3V = 2.53mA (ON), 2.41 mA (OFF) / 2.45mA (ON), 2.35 mA (OFF)
5V = 2.57mA (ON), 2.45 mA (OFF) / 2.8mA (ON), 2.94 mA (OFF)
12V = 2.88mA (ON), 3.03 mA (OFF) / 2.59mA (ON), 2.47 mA (OFF)

ohsix, 2017-06-27:
got a lot more of these now, can check them all for variance, will re-open if the results are significant

closed 2017-06-27 by ohsix

Joerg-rw, 2019-04-07:
confirmed.
Sidenote: a superbright blue LED works just great when connected to OUT and GND, no series R, consumption of complete circuit not changing noticeably (my DMM just still shows 0.002 / 0.003A)

dianarodriguesv, 2019-06-19:
Hello @hurvajs , could you share how you performed the measurement? I was acquiring with a data logger, and I was increasing Vin from 5V to 12V to check input consumed current, the current went from 2.8 up to 6,7mA,