jdesbonnet/RCWL-0516 #3: R-CDS operation inverted?

JeffreyRodriguez, 2017-05-22:
From the Readme:

> R-CDS	the VCC is in parrel connection with CDS(RCWL-9196 pin 9) through R-CDS. Connect the LDR at the R-CDS to turn off the detecting function at night. (?? TODO: make sense of this)

Perhaps this is setup to turn the detector *on* at night? I have a few PIR-activated LEDs around, and they spend the day charging.

I could be wrong here, still waiting for my modules to arrive.

dhharker, 2018-01-25:
Yes I think you are right. I have added an explanation in #19.

xuancong84, 2020-11-05:
My understanding of the table row

CDS | LDR 10-20k RL, U_LDR > 0.7V = On
-- | --

and PCB labels is that for CDS and R-CDS, one is for adjusting the luminance threshold for triggering output (I guess it's CDS), the other is for inverting luminance response (whether to trigger in brightness or in darkness) (I guess it's R-CDS). However, since the 2 resistors have one common pin, they will interfere with one another. Not sure whether this is correct.

Joerg-rw, 2021-02-28:
mounting a LDR in R18-CDS will cause enabled-in-dark
mounting a LDR in R16-R-CDS and a sufficiently small pulldown R (say, 1kR, or a 10k brightness adjust trimmer) in R18 will cause enabled-in-light.

with LDR in R18-CDS you may solder a 4.7MR brightness adjust trimmer to R16-R-CDS