jdesbonnet/RCWL-0516 #6: Info: 3.3v output PIN

itsjustvenky, 2017-07-19:
I thought that 3.3v output PIN cab be used to drive some other sensor or something like that,but it doesn't have enough juice to drive sensors.
So I put the pin to good use by connecting to GPIO of ESP and detecting the failure of the sensor. Incase if the sensor is not working or failed for some reason, we will have no power at 3.3v pin.

Thoughts about this ??

jdesbonnet, 2017-07-19:
It would be interesting to see how much current can be drawn from it. How
much current did you need as a matter of interest? Joe.

barewires, 2017-08-02:
I put a 68 ohm load resistor on the 3V3 voltage regulator output and see no detrimental effects at 5 volt and 10 volt Vin and draws 52 mA.
On another observation, plugging the sensor directly into a small breadboard with a right-angle header, the mass of metal caused unreliable operation.
Best of all is running a RaspberryPi Zero W (for Wifi) and Node-RED on autostart, powered from an Easy-Acc Power Bank, it logs the time and date in a file and emails and tweets me at a set interval.

barewires, 2017-08-09:
Testing the OUT pin with a full range of LED wired direct to ground with no dropping resistor shows very low current limiting from 0.20 - 1.10 mA, 3-5-10 mm, opaque, Water-Clear, WC-ultra-bright, VF 2.09 - 3.39 volts.  Full spectrum red, orange, yellow, green, blue, white and UV tested. LED typically run at 20mA so a transistor and resistor to 3V3 voltage regulator output would illuminate at full brightness.

The 3V3 output at 100 mA is by far the cheapest way to power an eForth STM8103xxx minimum dev board or equal, like the tail wagging the dog.   
https://github.com/TG9541/stm8ef/wiki

ortegafernando, 2017-08-09:
@barewires Do you think that tue OUT pin will be enought to drive a transistor ? And then, that transistor drives a relay with the 3.3V output pin ?

barewires, 2017-08-10:
Thanks for pointing this out, I have corrected it.

ortegafernando, 2017-12-01:
Does somebody check this? I am using the 3.3v pin output to power my ATTINY4313 and it doesn't work well, I think it is the 3.3v regulator of the RCWL-0516.

Any experience powering a IC with this pin?

regards.

barewires, 2017-12-02:
Did you read the read the above posts showing 3V3 works up to the rated specs of 100 mA. What is your VIN voltage?

ortegafernando, 2017-12-02:
Yes, I see them. My VIN is 5V from a powerbank and I have tried also with a 1000uF capacitor.

I will try with 12V with and without capacitor and I will write again. Thanks.

barewires, 2017-12-02:
OK I know what is happening; the powerbank cannot work with such a low output current of 3 mA, it will be on for a while and then will automatically shutdown as it thinks the 'phone' is fully charged. You don't need any caps. I have tested 4 to 28 volts VIN.

ortegafernando, 2017-12-02:
Thanks for your answer but this is not the problem. I have NRF24 (3.3v), ATTINY4313 (3.3v) and RCWL-0516 (5V), I have also some leds in the ATTINY4313 to show me when NRF24 is connected ok, and others leds to see if my sketch is correct (I have also used serial output of the ATTINY4313 but in this case I have to avoid my NRF24 code because of memory size). 

So I am sure that my powerbank is not shutting down. I will take a 12v battery and check again and I will try to see something with my voltmeter in the 3.3v pin (maybe I can't see anything).

Thanks.

barewires, 2017-12-03:
That is interesting, all of my powerbanks - Easy-Acc, different sizes even a solar USB charger shut down with a single RCWL-0516. How much current do you use with the entire setup? All of the devices are very low current, NRF24 - 15 mA, RCWL-0514 - 4 mA, still I think of powerbank shutdown. How is it with 12 volts going to VIN?

ortegafernando, 2017-12-05:
Hi, I have just done my trials.

I am working with a breadboard

3.3V rail has: 3uf electrolitic, 1uf tantalum and, 0.1uf ceramic
5.0V rail has also the same

3.3v rail powers nrf24 and attiny4313
5v rail powers RCWL-0516

NO LEDS now, i get managed to get some serial debug code and all nrf24 stuff.

RF24 has also a 0,1uf and a 47uf capacitors.

CHECKS:
1. from PC's usb port to arduino uno (only to give me 5V and 3.3V voltage outputs), and its 5v and 3.3v pinout to each rail: OK, everything is working
2. from PC's usb port to arduino uno, which gives me 5V. Then RCWL-0516 will give me the 3.3V voltage connecting its 3.3v output pin to 3.3v rail: nrf24 and RCWL-0516 NOT working
3. 12v from power adaptor to RCWL-0516 and RCWL-0516's 3.3v output pin to 3.3v rail: rf24 woring, RCWL-0516 NOT, not stable in general.
4. 12v with 1000uf capacitor to RCWL-0516 and its 3.3v output pin to 3.3v rail: rf24 woring, RCWL-0516 NOT working.

I have also tried another rcwl5016 unit with same results.

And two last checks:
Powerbank 5v to RCWL-0516 and it's 3.3v output pin to 3.3v rail: nrf24 and RCWL-0516 NOT working
powerbank 5v to RCWL-0516, DIODE LED from 5v rail to 3.3v rail (in order to get about 2.7-3.3v): nrf24 and RCWL-0516 NOT working

So is solution to power 3.3v from another source and not from RCWL-0516, isnt it ?

Thanks a lot.

NRF24 will consume not more than 12-15mA, but may be there are some peaks, or whatever.

barewires, 2017-12-05:
I would power the uno from host USB, uno 5v to RCWL-0516 VIN, uno 3.3v to NRF24 (3.3v) and ATTINY4313 (3.3v). I would also get rid of all of the caps, just causes needless complexity. If power or reliability problems occur later then reconsider installing them. All (most?) boards already have filter and decoupling caps installed.

If you read the other issues, then by my posts it is recommended to keep all RF well away from breadboards and all metal. A breadboard is a mass of metal contacts hidden below a plastic grid and will cause all sorts of problems if plugged directly in. Put the RCWL-0516 on 3 wires, 5 cm long and keep it well away from the breadboard.

With only VIN, GND and OUT connected to a meter, do you see it responding?

My results with powerbanks and a single sensor always fail after a few seconds as the power shuts down.

I can run a PI Zero W remotely for 4 days at a time (Easy-ACC powerbank) with the cat detector emailing or tweeting me running Node-RED on Raspbian. This allows me to remotely log in on Firefox 192.168.1.18:1880 and make changes to the GUI. Using SSH I log in from my tab or phone and view the text log files. Worth considering a Pi Zero W, £9.60 and an SD, and a 50p sensor.

ortegafernando, 2017-12-05:
> I would power the uno from host USB, uno 5v to RCWL-0516 VIN, uno 3.3v to NRF24 (3.3v) and ATTINY4313 (3.3v). I would also get rid of all of the caps, just causes needless complexity. If power or [...]

I have already done it. This way it works perfect. But if I power NRF24 and ATTINY4313 from the 3.3v pin of the RCWL-0516 then nothing works

May be the internal LDO regulator of theRCWL-0516 is not good enough to power an ATTINY4313 and a NRF24 (this has peaks while transmitting, may be this is the main reason not to work properly)

barewires, 2017-12-08:
Exactly how much current is being drawn by each device and what are the high current loads (LED etc) attached to all devices? I have confirmed at the top of this post a full 100 mA load on the sensor, but I didn't vary the load.

I have been running eForth on a C0135 4 x relay board and have done a slow random select of the 4 relays, so I will wire a combination of power resistors to test the varying loads on the sensor.  
https://github.com/TG9541/stm8ef/wiki/Board-C0135

ortegafernando, 2018-01-09:
Hi, Probably it is not a problem of how much current, probably it is about stability of the 3.3V regulator of the RCWL-5016 board, isn't it ? ATTINY413 will be about 5-7 mA powered with 3.3V , NRF24 has 12mA in transmission, <1mA in stand by; and nothing more.

pokergeist, 2019-08-04:
Here's another data point regarding using the 3V3 line to power a micro-controller as was suggested on another forum.

I'm controlling an RGB LED "Neon" strip with a 3V ItsyBitsy micro-controller. 12VDC powers the RCWL board (Vin) and the anode of the LED strip. I use the RCWL board's 3V3 output to power the ItsyBitsy. The RCWL OUT pin is connected to a digital input line on the uC (interrupt and level polling for retriggering). This triggers full power white task lighting via low-side n-channel MOSFETs for the red, green, and blue channels. Once OUT drops to a low level, the uC waits for 90 seconds before switching to a low power "effects" mode (color morphing, whatever).

The problem occurs when the uC enters effects mode and applies pulse width modulation (PWM) to the MOSFET gate inputs. The RCWL immediately triggers. I can ignore the OUT signal until the fade is done plus enough delay for OUT to drop. But within an hour or two the OUT pin is locked at a high level.

Using a PicoScope, the 3.3 signal looked good, I also drove another I/O line to mark where my code was executing and confirmed that any of the (3) analogWrite() [PWM] calls triggered the OUT line. Almost any value for analogWrite(0 to 255) except 0 and 1 cause OUT to trigger. I did not measure the current draw from the 3.3 line.

The MOD: I disconnected the 12V feed going to the RCWL Vin pin. Then I added a USB power supply to the ItsyBitsy, which now backfeeds 3V3 to the RCWL board via the 3V3 line. Everything works as expected now.

dlin4668, 2021-03-12:
Not 100% sure this post is related to the OUT pin or the 3V3 pin. 

Anyway, I have just confirmed some results that has menionted by **barewires**: the current flowing from OUT pin is very small:
1. If I connect OUT to a limiting resistor and then a low Vf LED, there is no light
2. If I connect OUT to the low Vf LED directly, the light is very dim (some higher Vf LED, no light at all)
3. If I measure the voltage of OUT (i.e. Vf), it is only 1.82V
4. This has been tried on my 5 boards.

When I remove the LED, the OUT becomes normal at 3.3V. 

Conclusion: The OUT pin is high at 3.3V (after detecting a motion), but its current is very small, so OUT(Vf)  is about 1.82V in my case.