jdesbonnet/RCWL-0516 #7: Variable Output

BiatuAutMiahn, 2017-09-03:
I was wondering if there were a way to modify this to give a variable or PWM output representing a distance, or speed?

I figure a number of these placed strategically might allow one to map and object in 3d space.

barewires, 2017-09-03:
The device is omni-directional meaning it sees in all directions - XYZ, therefore making it useless as a focused sensor. In my cluttered livingroom / lab I get 3 m response, any movement whatsoever triggers it. It sees through drywall and even people outside my front door when it is placed in my hallway.

BiatuAutMiahn, 2017-09-04:
ok, but if you were to put a few like 3 or 4 of these would you be able to detect the position of a moving object? like GPS or triangulation?

barewires, 2017-09-04:
I have experimented with multiple units in close proximity 1 cm apart, pointing different directions and they all react to any motion at the same time. There is no possible way (without building microwave horns or wave guides) to get directional or distance information, as far as I can see.

BiatuAutMiahn, 2017-09-04:
I see, however theoretically, I wonder if one were to tweak the board to that the output signal 'on' time is reduced from a couple of seconds to like 10ms or around there, and then hook the outputs of 3 or 4 to the GPIOs of a raspberry Pi, or arduino...and set the modules further apart, like 1ft or 4 inches, there should be a very small time difference in which each module is being triggered.

barewires, 2017-09-09:
No, Not going to happen. They are analog devices and the timing is variable, controlled by op-amps, capacitors and resistors and physical placement and proximity to metal. While not suited for what you want they are incredibly amazing devices for detecting cats and lions coming in and out. Bon chance!

barewires, 2017-09-12:
As an observation I put the device in a round metal 9 mm can with a 7 mm plastic window (fidget spinner) and get good directional response through the window.

BiatuAutMiahn, 2017-09-12:
hmm nice. Ty

nissimzur, 2017-09-17:
Does someone did PCB layout and can share or sell it to me? 
nissim.zur@gmail.com

Joerg-rw, 2021-02-28:
> set the modules further apart, like 1ft or 4 inches, there should be a very small time difference in which each module is being triggered.

Those devices have operation principle immanent interference between them, you basically can't operate them in a meaningful non-interfering way within same range zone. Device A won't be able to tell apart reflections of own signal from those of device B, C...

I did tests and been able to trigger a sensor from simply powering up another one in 6m distance and that wasn't the distance limit but simply the max distance I could test in by scenario.
Considering how much stronger direct signal from antenna of B to antenna of A must be, compared to signal from A getting reflected and received again on A, I'd expect you need at least 10m distance between sensors to have sufficient attenuation of crosstalk

xenoeng, 2022-04-12:
You could only power and check each one in turn, cycling through them? Trickier to do with a distributed system but not impossible. ESPs and MQTT could provide the messaging and sequencing needed.

Joerg-rw, 2022-04-12:
I just an hour ago thought about some prerequisites for that, since the plain device as is needs some seconds to power-up, but I guess that's mainly the opamp and timer stuff in the chip. The RF-section probably could get enabled and stopped within microseconds, you just need to avoid triggering a false positive on the chip then from that.
I'll report if I actually look into that any closer

xenoeng, 2022-04-12:
Any false triggering at power up could be ignored by the software. Could have a configurable settling time for different environments and conditions.

Joerg-rw, 2022-04-12:
sure, no problem technically. Just it reduces the poll time so much than you can't do reasonable polling since all sensors of a RF-zone have to be down for too long when one of them just powers up. Long enough so somebody could walk through the complete rnage without getting detected

xenoeng, 2022-04-12:
Interesting! Can you share?
Watching with interest.