jdesbonnet/RCWL-0516 #8: Question: Pulling low on CDS will fully disable the module ?

itsjustvenky, 2017-09-11:
I have tested pulling LOW on CDS module won't trigger the motion, but I just know whether the module is fully disabled or just triggering part. I kept the module close to my desktop and so I am worried about sitting close to microwave radiation. Any thoughts ?

barewires, 2017-09-12:
I just inhibited the CDS by grounding but the the current drawn stays the same at 3.15 mA. It would be best to switch the VIN with a GPIO (at 5v) or a transistor and wait for the device to settle.

itsjustvenky, 2017-09-17:
@barewires Thank you so much for the info. I thought of connecting ESP8266 GPIO directly to 3.3v pin to turn module ON/OFF. The RCWL module required 3mA max for operating, so the GPIO can provide sufficient current. Thoughts ??

Another thing, I read through webpages and I found 20mW/Cm2 is the max allowed according to standards. Thoughts on this ??

Thanks

barewires, 2017-09-17:
The 3V3 pin is a voltage regulated OUTPUT intended to supply external circuitry only.  I have tested it at full load of 100 mA and it only gets hot beyond specs above 15 VIN. The VIN is specified as 4-28 volts however 5 VIN is the minimum with a full load drawn on the 3V3 output. View other issues for further details. I will investigate the workings at 3.3 VIN as you have sparked my curiosity.
Check out my info at http://forum.hobbycomponents.com/viewtopic.php?f=73&t=2181&p=5642#p5642

itsjustvenky, 2017-09-18:
@barewires You are right. I just tested inputting 3.3v on 3.3v OUT pin and I get tons of false triggers. 
I get LOW/HIGH randomly on GPIO, I think the only option is to power with VIN with 5V.

barewires, 2017-09-18:
Yes, VIN is the only way to power the device, the specs say 4 to 28 volts and I just tested that. There is about a 10 second delay before the device works after powerup. This should be taken into account if you switch the VIN on. Below 4 volts VIN is unstable.

stonemull, 2017-10-18:
@itsjustvenky 

Microwave radiation is non ionising, it does not affect the cells .. if all the input power went to microwaves then (say 5v, 3mA = 15mW) then you are radiating 15mW ... your wifi router is many magnitudes greater as is the phone in your hand. Don't worry about the device, it is harmless.