jdesbonnet/RCWL-0516 #11: Question: Minimum range by R-GN adjustment?

iSuslov, 2017-10-16:
Hello, thank you very much for your research. How do you think, is it possible to reduce its range to 30cm or even less, by adding high resistance to R-GN?

barewires, 2017-10-16:
R-GN, Resistor Gain, is a feedback resistor added in parallel with installed 1 M ohm which will reduce the feedback of the 1st stage OP-AMP. The circuit for the BISS0001 chip is almost identical with exceptions of the added 3V3 3.3volt regulator output and the removal of the non-retriggerable feature.
I will install a 1 M ohm 10-turn potentiometer, set initially to 50% and experiment with the function. I have two in my open side window, to detect my neighbours cat, and one with a 1 M in parallel is still too sensitive as they see me typing and mousing 2 m away.

dibs, 2018-05-26:
@barewires Did you manage to get some results? I am trying to limit range to maybe 2m if possible.
Any tips woould be appreciated.

iSuslov, 2018-05-26:
@dibs https://www.youtube.com/watch?v=YLMlRhxgTWQ watch this video from 4min. It is on russian language but briefly this is what happens: 
Guys said that he tried to adjust range with R-GN and failed to get desired results - range was limited but effective range was from 0.5m - 3m, so he did not want to exclude first 0.5m. 
So he started to experiment with antenna resistor on the other side of this board, and got accurate and  desired results replacing 220 ohm resistor with.. see the video(first use adjustable resistance, find your desired resistance and replace)

dibs, 2018-05-29:
@iSuslov Wow, this is good news. I will attempt this tuning and let you all know how I get on.
Sadly I can't understand much Russian though..

iSuslov, 2018-05-29:
@dibs it is not necessary to understand what he is saying. Just look what he points to. I briefly explained the context.

bachoo786, 2019-01-08:
Hello @iSuslov 

I want to achieve a 30cm distance of detection range. What would be the suitable resistor to connect? Many thanks.

iSuslov, 2019-01-09:
@bachoo786 I don't know man. Take a variable resistor, solder it instead of antenna resistor and find the right for you. I did not experiment on that yet.

bachoo786, 2019-01-09:
@iSuslov thanks. 

Do I solder it with the R-GN?

iSuslov, 2019-01-09:
@bachoo786 no, see my https://github.com/jdesbonnet/RCWL-0516/issues/11#issuecomment-392249631  why you should not use R-GN spot. You should solder it instead of antenna resistor like on the picture here https://github.com/jdesbonnet/RCWL-0516/issues/11#issuecomment-392770103 and empirically find the right resistance for your range.

To be clear, instead of this resistor:

[image]

bachoo786, 2019-01-09:
@isuslov so I remove the 221 ohm resistor and solder the variable resistor is that correct?

iSuslov, 2019-01-09:
@bachoo786 yes. And that adjust resistance. If you don't want to use variable resistor (`trimmer resistor`) you can try solder 80 Ohm, it seems to change range to 30-40 cm according to https://github.com/jdesbonnet/RCWL-0516/issues/22#issuecomment-395128341

bachoo786, 2019-01-09:
@isuslov thanks. I have a variable resistor and I can experiment it. However my variable resistor has 3 pins can you please show me where do I solder them on the sensor? I know the pins are vcc, gnd and data.

bachoo786, 2019-01-09:
@isuslov as per your photo you have not used pin 1 which is the ccw. I thought you should be using pin 1 ccw and pin 3 cw to measure the values but you are not doing that as per your photo?

iSuslov, 2019-01-10:
@bachoo786 good question. It is not my photo, I used it from youtube video and yes, seems that on this screenshot the guy uses trimmer resistor as a voltage divider. I'm also confused. I think it is better to try it yourself, I'm sorry I will be in my small lab only in two weeks. Could you please try different approaches and let me know what happend.

bachoo786, 2019-01-18:
@iSuslov sorry I have been away and back now.

I tried to solder the variable resistor and set it to 80ohms but the results are very flaky i.e. sometimes the sensor detects motion and sometimes it doesnt. 

I am thinking of connecting the variable resistor across the data pin and ground in the main terminals near 3v3 and VIN. do you think that would work?

iSuslov, 2019-01-18:
@bachoo786 Hi, actually I also tried this way and got similar results. It detects only fast motion and I don't like that. I would go with combination of soldered variable resistor PLUS variable resistor on R-GN soldering spot. Maybe experimenting with that will give some better results. I don't think manipulations with **data** pin is a good idea.

mujahed1987, 2019-03-25:
> @dibs https://www.youtube.com/watch?v=YLMlRhxgTWQ watch this video from 4min. It is on russian language but briefly this is what happens:
> Guy said that he tried to adjust range with R-GN and failed to get desired results - range was limited but effective range was from 0.5m - 3m, so he did not want to exclude first 0.5m.
> So he started to experiment with antenna resistor on the other side of this board, and got accurate and desired results replacing 220 ohm resistor with.. see the video(first use adjustable resistance, find your desired range and replace with static resistance).
> 
> Finally: now he claims he can adjust from 5cm to max range, playing with antenna resistor. And since he also needed 2m range, his choice for antenna resistance appeared to be 165ohm

Hi,
I tested this solution with about 10 Sensors but to of them was defected after a while, I think this workaround solution could introduce issues by affecting [ Impedance matching](https://en.wikipedia.org/wiki/Impedance_matching) between the antenna and the entire circuit which may bring unwanted signals and powers to the circuit. I still doing my experiments, and I need confirmation from others if any body else had the same issue. I want you to know that I am working in real environment where the sensor installed with other component to automate the lights switching in our company,  reducing the range of detection and keeping high sensitivity is very important for our application. so I will keep doing tests to find best solution.
Good luck everybody

ismailtan, 2019-05-04:
I just received 8 of these sensors and I'm trying to reduce the sensing distance as well. I want to use these sensors to make 6 installations using 1 sensor on each installation. I need a distance of 2 meters (or smaller) between the installations.

Sensing distance out-of-the-box for me is around 3 meters. Ideally I would love to reduce the sensing distance up to 2 meters (or less). By adding a 1M ohm resistor on R-GN I'm able to reduce the distance with approx 40cm.

Adding smaller resistors are resulting in no triggers on pin-out and adding a couple of 1M ohm resistors in series (because of not having bigger resistors then 1M ohm) seems not to help reducing the sensing distance. The sensing distance is the same as just adding 1x 1M ohm resitor on R-GN.

Anyone figured out how to reduce the distance? Or has any other suggestions? I did not experiment with the combination of replacing R9 with a trimmer pot and adding a trimmer pot on R-GN. I'll report back when I tried this .

djay07, 2019-06-02:
Can anyone recommend a resistor to limit the distance to max 2 meters?
I am new to this and trying to make it work. 
My sensor freaks completely out, since it even seems to "see" movement of my neighbors.

turbouk2312, 2020-05-06:
Sorry to bump this old thread, but did anyone get anywhere with limiting the range? I've tried a few resistors but I cant get anything sensible from the sensor after soldering them to R-GN pads. Ideally I'm after about 50cm-1m range.

ismailtan, 2020-05-21:
Just wanted to give it another try today and I figured out something promising.

Placing a 220k ohm resistor on the R-GN pads gave me a max range of around 10cm or so and a 270k ohm resistor around 2 meters.

I tried to cut the max range in half by placing a resistor value of 245k ohm (1M ohm in parallel with 330k ohm in lack of having a 245k ohm resitor), but this also resulted in a max range around 2 meters. I'm guessing I have to ramp up from 220k ohm in smaller steps.

Will post more when I get the chance to thinker around more.

JohaGit, 2021-02-02:
[image]

JohaGit, 2021-02-02:
[image]

dlin4668, 2021-03-06:
@JohaGit I would like to learn how to understand this chart better. 

1) Are these 5 points on the chart from your experimental results? 
2) By putting different R-GN values parallel with R4 = 1M, the sensing distance will work from 0 (0.10Mohm) to 7 m (about 5Mohm)? 
3) You are not suggesting to change R9 (220hm) at all? 

Thank you!

JohaGit, 2021-03-07:
> @JohaGit I would like to learn how to understand this chart better.
>
> 1. Are these 5 points on the chart from your experimental results?
> 2. By putting different R-GN values parallel with R4 = 1M, the [...]
Note: I inserted two pics in my original post to ensure that everyone are aligned in the resistors we are referring to. Because there are unclarity  in some other posts:
Adressing your questions:

1. Yes. Note that this is a radial “free” field. **Any** object in the vicinity like walls, carpets, metal cups, porcelain or red paint with iron/oxide will have various degrees of reflection, interference or attenuation.  All in all I took about 40 readings with different scenarios but used this basic one since it matched reality with less modifications on the circuitboard.
2. Yes. You will see that the y axis is on the logarithmic scale. That means that there is extreme nonlinearity at small distances and quasi linearity at the furthest distance. Also, the red area is indicative of polar coordinate sensitivity around the receiver at distances 0-30cm. This is caused by the dimensions of the metallic on circuit board itself starting to have a role on the sensitivity==> within that red area it makes a difference whether you are directly (in a plane) in front / behind the face of the (0 ° radians) circuitboard, or you movement is skewed with relation the plane of the circuitboard.

3. I tried it with both. But found it sufficient for my purposes not to change R9 at 220 Ω. Since this is a high frequency colpitts oscillator circuit, changing R9 affects the relative logarithmic sensitivity linearity within the reception range. Thats why playing with R9 values “bulge” sensitivity to the left or right within the frequency window.

dlin4668, 2021-03-08:
@JohaGit 

Thank you for your profound answers with the professional lab results. 

I am asking my hardware colleague to modify the modules with either R3/R-GN or R9 solution before shipping, so I would like to be sure R3/R-GN installation is adequate.

Sorry for another questions: 
1. Could you comment about the first 0.5m problem by iSuslov about the video "_he tried to adjust range with R-GN and failed to get desired results - range was limited but effective range was from 0.5m - 3m, so he did not want to **exclude first 0.5m**_"
2. On your schematic, I cannot be sure your R4 value is also 1M Ω and cannot find the 3.3V LDO either. Is your R4, that is untouched, 1MΩ like mine? (https://raw.githubusercontent.com/jdesbonnet/RCWL-0516/master/doc/john_taylor_rcwl-0516-0.png)

JohaGit, 2021-03-08:
Sure. 
1. iSuslov Referred to the russian youtube video where the guy went and modified that 220 ohm R9 “dirty Harry” trick. And that is how electronics rewards him: by creating „blind“ spots in his case. No, the red area that I am referring to is that you can anyway not assume uniform radial sensitivity  ina sphere around the board. Imagine that radial map as a 3D blob around the circuit for small distances.
2. Yes my R4 is also 1M ohm. If you cannot see that chip next to R-GN it means you likely have the older board so your 3.3V regulation is not that good. That can cause some further quality issues but not a show stopper.
3. No. A trimmer is ideal for calibration.a)  Decide first what range you are interested in.  b) Then lookup from the natural logarithm graph what the theoretical resistor should be say X ohm. c) choose a  log potentiometer double that:  ie 2X and a linear potentiometer in series which is ca. 10% of X (0,1 * X) for fine tuning. Connect them in series before soldering and calibrate them to X. Then solder them on without touching the knobs d) place the receiver in the environment where it should be. If indoors, the sensitivity is higher due to many reflections and interference patterns e) if you change the resistor values by turning the knob, start with the fine tuning first to calculate the range. Detecting a human is different than a dog or a moving metal part like a factory line with metal parts on it.  remember to move your body away because your body is one big fat aerial with all that water content. If the calibration is done and you are happy and you intend to make it for multiple boards then replace it with 1% resistors. You can even fine tune it further to millimeters by scraping of carbon of a fixed resistor but that is a overkill for such a cheap chip. For a part number just type „log pot 1M“ in ebay and you can select all the variants there, for your closest match.

JohaGit, 2021-03-09:
Hi, I typed a 2 page response, maybe you can ask the admin (jdesbonnet) of this thread to undelete my last comment for you

dlin4668, 2021-03-09:
@JohaGit 

Acknowledged. I will install the R-GN with a suitable trimmer resistor and keep the original R4. Thank you!

dlin4668, 2021-03-09:
@jdesbonnet I can receive some contents from JohaGit in email but cannot see those contents (and 2-page response) on this page. My email address is daniel.lin0913@gmail.com, if this helps.

JohaGit, 2021-03-09:
> @jdesbonnet I can receive some contents from JohaGit in email but cannot see those contents (and 2-page response) on this page. My email address is [daniel.lin0913@gmail.com](mailto:daniel.lin0913@gmail.com), if this helps.

Hi daniel, for the sake of efficiency, just capture my post in your email as quote and include it in an additional post of yours, so that it can then be reused by other members also interested on the answer...