jdesbonnet/RCWL-0516 #14: Sensitivity to interference

aholatom, 2017-11-26:
My motion detector was triggering a lot of false alarms.  I suspect the nearby WiFi router triggered these.  I tried to make a sheet metal noise shielding box for the radar.  However, it stopped working.  I did leave about 10mm space on the bottom side.  It is not sensitive to movement any more.  But it does trigger sometimes when slightly moving the device, maybe my power supply clips rattle, or some electro-acoustic phenomenon.  Any other ideas how to reject interference?  I wanted this as small as possible as it is inside a raspberry-pi box,

[image]

barewires, 2017-11-26:
The sensor is sensitive in all directions (omni-directional) favouring the transistor side as shown but any metal will cause problems including metal boxes, large electrolytic capacitors, ground planes, metal shields and huge transformers. All one needs is three wires 5 cm long and a plastic box.

aholatom, 2017-11-26:
Yes, well understood.  I did anticipate some effect of the metal box, but not a complete failure to detect anything even at a very close proximity.  I did try to put it the other way around in the box with same bad results.  I just wanted to report my findings and save somebody else from the trouble making a metal box.  And it does not work in a plastic box at my location due to the WiFi router in the same small room.

I now experimented some more and it seems **it works** when I remove the metal covering the chip.  A bit surprising as that is the low frequency part.  I will try to put it in the Raspberry Pi enclosure and test it with the WiFi router next.

barewires, 2017-11-28:
Looks like a well engineered massive failure with the best of intentions, a noble effort. This forum is at the bleeding edge of this new technology. I think that we all can now agree on no metal at all around the sensor. Let's get some input on feed horns and trying to focus the spherical radiation pattern in a specific direction.

aholatom, 2017-11-29:
With some additional tricks I got it now working with the help of the metal shielding.  No more false triggers from WiFi interference.  Thanks for the input!

ortegafernando, 2018-01-09:
Hi @aholatom  please, share your experiencia not to have more false triggers, thanks.

karanmakharia, 2018-08-30:
@aholatom How did you get it working?

capitalfuse, 2018-09-01:
Try to twist 3 wires into one line to resist the WiFi noise.

YasirShahahzad, 2018-09-08:
How did you solve the problem to get it work??

langeludo, 2018-09-09:
It seems that ferrite ring core helps a lot in shielding external radio wave interferences.

nimamohammadi, 2021-10-03:
Hi,
I had a lot false trigger with this module.
1- first keep this module away from WIFI signal.(in my experience with ESP8266, 15-20 cm is good)
2- place a 100nf cap between output and GND of RCWL.

hope it work for you