jdesbonnet/RCWL-0516 #18: Supplying with 3V3

PopoTheSapien, 2018-01-10:
Hi.  Cool sensor.  This is not an issue, but is it possible to power the radar from 3v3 without using a step up?

DurandA, 2018-01-18:
According to [this answer](https://electronics.stackexchange.com/a/298590/80406) and the schematic, powering the RCWL-0516 directly from the 3.3v pin seems to work fine. I have yet to test it though.
Please give us some feedback if you try it.

DurandA, 2018-01-18:
On the other hand, [this comment](https://github.com/jdesbonnet/RCWL-0516/issues/8#issuecomment-330140182) indicates that the module trigger false positives when powered from 3.3v pin. It could also be a consequence of a noisy power source.

barewires, 2018-02-01:
Just finally put 3.3 v on the 3V3 output and it appears to work by back-feeding the device. I got a detection at 4 m through a doorway into my kitchen. This is not in the specs but it is worth trying with caution.

PopoTheSapien, 2018-02-04:
Interesting.  I will give it a try too.  Would make it much easier to work with.  Will post back results on some devices I have running.

od1, 2018-03-23:
Did some quick tests feeding clean 3.3V on the 3.3V pin, then with 5V on the Vin. The RCWL-0516 appeared to behaved exactly the same. Pin Out was 3.3V when I moved in front of it at about the same distance. No false trigger detected.

m-anish, 2018-05-27:
For reference, giving it 3.2V makes the sensor malfunction - I was trying to run this directly off of an 18650 cell which was partially discharged and at 3.2V and the module kept turning the LED on and off.

Adding a boost module and running it on 4.0V seems to have solved the problem.

renne, 2018-06-27:
I connected the 3.3V output of the RCWL-0516 module with the 3.3V output of a [H801 LED controller](https://github.com/arendst/Sonoff-Tasmota/wiki/H801). Back-powering the RCWL-0516 with the 3.3V of the H801 works fine.

waldorffsmama, 2019-02-19:
My own experience: runs well on 3.3V until initializing the wifi on an ESP32. Once the wifi was on, it gave constant false positives. Changing to VIN with 5V sorted the issue and it ran with wifi on as reliable as before with 3.3V

kbhuinfo, 2019-07-18:
> My own experience: runs well on 3.3V until initializing the wifi on an ESP32. Once the wifi was on, it gave constant false positives. Changing to VIN with 5V sorted the issue and it ran with wifi on as reliable as before with 3.3V

I can confirm this. Same results here.

DurandA, 2019-07-18:
> My own experience: runs well on 3.3V until initializing the wifi on an ESP32. Once the wifi was on, it gave constant false positives. Changing to VIN with 5V sorted the issue and it ran with wifi on as reliable as before with 3.3V

Did you use a decoupling capacitor?

s0170071, 2020-01-06:
My tests show simultaneous false positives on all devices with the same supply of 3.3V from an Esp8266.
I assume that the Chinese put that 5V regulator there for a reason, especially on such a super low cost module...

hazymat, 2020-11-20:
For reference I've been running this directly from the logic level connections of a Sonoff which supplies 3.3 or slightly higher (3.34 I think) and it has worked reliably for the last few days. I have tested this on the bench next to the SONOFF (esp8266) and with the sensor wired up about 6m away from the microcontroller (Sonoff basic R2). I may install a boost converter to be safe though as I will be installing this into a ceiling.