jdesbonnet/RCWL-0516 #22: Adapting Sensor Range

Xpcker, 2018-02-15:
Hello everyone, 

Does anyone know if maybe modifying the voltage or resistance at some point could potentially adapt the range of the sensor? i made a light with the sensor but it turns on way too much.

Thanks!

Xpcker, 2018-02-15:
sorry just found this:

R-GN | The default detection range is 7m, adding a 1M resistor reduces it to 5m

Ill try to make my own PCB and try different resistors and post the result

mrmate2, 2018-06-06:
To have opportuniyy to adjust range of sensitivity( even starting from 5 centimeters!!!), you can replace R9 resistor .
Factory parameters- 220 ohm.
Reducing that valuehwill reduce range.
For example: 80 ohm gives maximum 30-40 centimeters sensitivity, as per my observations...
You can rplace R9 with adjustable resistor.
P.S.Sorry for my written English, I am from Ukraine.

vincegre, 2018-06-14:
```
To have opportuniyy to adjust range of sensitivity( even starting from 5 centimeters!!!), you can replace R9 resistor .
Factory parameters- 220 ohm.
Reducing that value will reduce range.
For example: 80 ohm gives maximum 30-40 centimeters sensitivity, as per my observations...
You can replace R9 with adjustable resistor.
```
Why don't you use the R-GN connection points on back done for that ? lot easier than changing R9 resistor !!

underwoodblog, 2018-07-08:
How can I go in the centimeter range with R-GN? Bigger or smaller than 1M?

jonhanford, 2018-11-21:
Definitely smaller than 1megΩ

I used a 220kΩ to cut range to 3.5m

id try a 47kΩ to get it to under a meter. good luck let us know.

kutirshilpi, 2018-11-27:
Hello, does anyone know how to change the triggering threshold? For example if I want to configure the device to output high if it detects mm range displacement? This question is not about range but triggering threshold level.

jonhanford, 2018-11-27:
Well - it's not such a sophisticated device that there's any kind of range calculation - what it's detecting is frequency changes in reflected radio waves caused by motion of the reflector. 

Thus, the triggering threshold is actually measured in terms of doppler shift, and change in the threshold would change the *speed* of the detected object required for trigger, but would have no effect on *displacement distance* (other than, perhaps, meters of displacement per second). It has no means of measuring any kind of distance, or being triggered by a change at a set difference. It's only "measuring" interference from reflections and doppler effect that are caused when something near the sensor changes position at above a certain rate. 

> The low doppler frequency difference is extracted by a low pass RC filter (C9 = 1nF, R3 = 1k, fc = 1/2πRC ≈ 160kHz) and amplified by the RCWL-9196 IC and treated exactly the same as a signal from a PIR sensor.

Theoretically if you changed out the components of the low-pass filter you could change what shift in frequency would be required to trigger sensor, thus changing what speed would be required for triggering sensor, but it sounds like for your application, you need something more of a rangefinder.

kutirshilpi, 2018-11-28:
@jonhanford thank you for your explanation.  So, if I put a 1000uF cap across I should be able to detect low frequency Doppler.   
`C9 = 1nF+1000uF, R3 = 1k, fc = 1/2πRC ≈ 0.16Hz` 

What if the motion is simple periodic , i.e., net Doppler shift is 0? I will try altering those and see the effect. 
I was not sure how mixing is happening with the RF echo.

mattdougan, 2018-12-24:
@johanford was that 220k r placed on the R-GN pads on the back of the board?

bachoo786, 2019-01-08:
@jonhanford are you saying to connect a 47kohm resistor to the R-GN or the R9?

And how to you use the R-GN ?

Thanks

iSuslov, 2019-01-09:
@vincegre thats why you don't use R-GN https://github.com/jdesbonnet/RCWL-0516/issues/11#issuecomment-392249631
@mattdougan @bachoo786 see explanation here https://github.com/jdesbonnet/RCWL-0516/issues/11#issuecomment-452764010

arihantdaga, 2019-03-02:
In my application i wanted to increase its sensitivity. I want to use it for general occupancy detection. Where if people are sitting in a conference room  and have very slight body movement, i want the sensor to be able to detect that. Right now its only able to detect big movements in body. WIll increasing voltage at Vin help in increasing sensitivity ? or decreasing The resistor value ?

barewires, 2019-03-10:
Just for the record I put a 'cat detector' in my upstairs neighbour's flat when she went away for a few days. It was connected to a Raspberry Pi Zero W (wireless) and actually emailed me Node-Red whenever I walked below it even through a concrete floor. Her next door neighbour and others in the common hall also triggered the device. So it's back to the old drawing board (how I miss those).

Joerg-rw, 2019-04-07:
> In my application i wanted to increase its sensitivity. I want to use it for general occupancy detection. Where if people are sitting in a conference room and have very slight body movement, i want [...]

I checked for VIN range: device starts operating at ~2.5V with poor sensitivity and very unstable and randomly ("noisy"), around 3.0V range and motion-speed sensitivity are almost normal, just a lot of erratic retriggers caused by end of a trigger period. over 3.0V device seems to operate normal, just like within specified range (4 - 28V)

Vin has no reasonable useful effects on sensitivity and range

Joerg-rw, 2019-04-07:
my device is **very** sensitive to motion-speed. Like 0.2m/s seems already enough to trigger it. I can't do one single normal move in my chair without the thing triggering a 2m away, not even smoking a cigarette

arihantdaga, 2019-04-07:
@Joerg-rw Thank you. But somehow my device is not that sensitive.. It makes me think..

thexperiments, 2019-08-10:
I currently setteled for 680k ohm on R-GN Pads, now nicely covers one small room and is not triggering on movements in neighburing rooms anymore.

naibaf7, 2019-12-03:
I used a 100 Ohm resistor in parallel to the 221 220 Ohm resistor and get a very consistent 30-50cm trigger range, perfect for hand-triggering light switches etc. by proximity.

miloit, 2020-03-07:
> I used a 100 Ohm resistor in parallel to the 221 220 Ohm resistor (resulting in approx. 68 Ohm resistance) and get a very consistent 20-40cm trigger range (the range varies a bit from chip to chip which I have modded), perfect for hand-triggering light switches etc. by proximity.
> What I noticed is that having a lot of other electronic devices such as a computer close by reduces reliability. Powering the sensor using a good 5V supply over an Arduino also works more reliably than using a computer to power it, which seems to introduce noise that again leads to false negatives (no trigger). Adding a capacitor to the power line of the sensor helps as well.

Have soldered the 100ohm in parrlel to the r9 ?? Or rgn??

deepaksisodia079, 2020-07-05:
I was able to modify the reange to few cms 4-5 cms with 40 ohm resistance put in R9.
I also tried multiple resistance with R-gn but was not successful.

BUT

I am into a new problem,before any modification or keeping the resistance to around 220 ohm on the R9 resistor keeps the current consumption of the chip around 2.3 mAh but with the 43 Ohm resistance the consumption increased to 12mAh.

I am planning to keep the current consumption to be very low sine i want to run the prototype on a 4 x AA Alkaline battery which has a current of 1500 mAH give or take and with this rate the battery would die in few days.

Can someone help me with this problem,i am a biginner with electronics as i am more of a computer engineer.

Can I put something else along with the resistor to keep the range withing 4-7 cms and also draw less current?
Or is there any other way to reduce the power consumption of the chip by removing some pins from the IC on the board.

Please help

BeitongT, 2020-08-17:
parallel the 220 ohm with 366.5 ohm from Potentiometer  get a perfect 1.5 m range.  I test the sensor with potentiometer on it and get 1.5 m range.  Then I cut the potentiometer off and measure it with my 35$ multimeter and get 366.5.

Joerg-rw, 2020-08-19:
adjusting R9 is a very poor idea as R9 adjusts the DC current trough the transistor. If anything at all, increase R9 to reduce both supply current and TX power. A better approach probably is increasing R11 or reducing R12.

Generally reducing range of device is a botch job as this is a transmitter and receiver detecting the reflected signal, the range pretty much depends on size and reflectivity of the detected object - plus, due to the Doppler operation principle, on speed of object towards/away relative to device. The latter is basically what counts most for R_GN. The lower R3/R_GN the higher the speed it needs to trigger.
The TX power only adjusts the (reflectivity * area)/distance^2 product the device can sense, in other words the larger and more reflective the object the higher the distance it still can get sensed at a specific TX power level. While you might be able to reduce sensitivity to a 10cm when waving your hand toward device, it will probably still trigger when you walk away from a distance where you could do that waving of your hand since your complete body has much larger area.

for short range proximity detectors, rather look into reflective IR sensors, capacitive sensors or the like, this device simply isn't made for this usecase and hardly can get adjusted to perform nicely for it.

deepaksisodia079, 2020-08-19:
> A better approach probably is increasing R11 or reducing R12.

I can try that too,but for now..i am taking input from pin 12 into micro controller and processing it to take a decision if the wave is with 10cms...i observed a pattern and currently it seems to be working but not 100% accurate...i still do get false triggers as the same wave pattern on plotter of arduino happens from some range.

Do you think increasing R11 could reduce the range?I can try that.

Joerg-rw, 2020-08-19:
Increasing R11 (or reducing R12) will lower the Base bias of the transistor, thus reduce current through Collector-Emitter and thus should massively reduce TX power and current consumption. It's probably very delicate an adjustment, with small changes of resistor causing huge effects, at a certain point

deepaksisodia079, 2020-08-19:
And I think the final register value would vary based on each RCWL-0516 unit.
But let me try and will post comments here.

Joerg-rw, 2020-08-19:
> I can try that too,but for now..i am taking input from pin 12 into micro controller and processing it to take a decision if the wave is with 10cms...i observed a pattern and currently it seems to be [...]

> i am taking input from pin 12 into micro controller and processing it to take a decision if the wave is with 10cms

btdt, with scope. It's quite difficult zo conclude anything meaningful from the signal shape. I sort of seen a discernible pattern for approach vs move_away, and even that is quite faint a difference in signal. Everything else is a mix of reflected signal amplitude and movement speed of reflector and pretty much nothing in signal shape tells about distance of reflector. When you think about it, it's not even possible by detection principle to decide the distance

Narcissus-GBC, 2021-02-27:
I am a student and doing a project. This is one of the sensors I have chosen and I have to limit it to 200 cm. The content on this page seems very useful, did you get the result?

Joerg-rw, 2021-02-28:
> I am a student and doing a project. This is one of the sensors I have chosen and I have to limit it to 200 cm. The content on this page seems very useful, did you get any result?

As said in my last post: the signal to exceed threshold is determined by (Area_of_object * "albedo" * speed_of_movement) / (distance * TX-power * sensitivity)
Pretty much any other sensor concept works better for exact determining of max range than this one

JohaGit, 2021-03-07:
> I am a student and doing a project. This is one of the sensors I have chosen and I have to limit it to 200 cm. The content on this page seems very useful, did you get any result?

Refer to my graph in another post in this repo: (ca. 68k + environment tuning)
https://github.com/jdesbonnet/RCWL-0516/issues/11#issuecomment-771810294

valioiv, 2021-10-02:
@all, what a potentiometer value I should put and which resistor I should replace to achieve adjustable range from 50cm to 5m..? Probably have to keep another resistor in series...

auto-the-automator, 2023-08-09:
For record, replacing the D9 resistor with another resistor (or a potentiometer), I could not get any of these resistor -stanza to work. 

By that I mean: yes, I was able to adjust the resistance, and yes, behaviour of the RCWL 0516 chip changed. First it seemed, that there was some luck limiting the range of the sensor, but few minutes of testing showed that chip just became prone to false negatives,  aimed at around 200cm range, but it DID detect movement also between 2m - 7 meters - but very unreliably. Tried with different chips, similar results, resistor mod just makes the chip unreliable.

My observations are more in line with what someone else has observed here:

https://community.home-assistant.io/t/rcwl-0516-microwave-radar-sensor-experiences/40887/22
https://community.home-assistant.io/t/rcwl-0516-microwave-radar-sensor-experiences/40887/25

Thanks

Joerg-rw, 2023-08-09:
s/D9/R9/
Yes, I already mentioned this in https://github.com/jdesbonnet/RCWL-0516/issues/22#issuecomment-676614390
Many thanks for the two links, HeyImAlex Alex knows his stuff, recommended read.

agenasrl, 2023-11-28:
> To have opportuniyy to adjust range of sensitivity( even starting from 5 centimeters!!!), you can replace R9 resistor . Factory parameters- 220 ohm. Reducing that value will reduce range. For example: 80 ohm gives maximum 30-40 centimeters sensitivity, as per my observations... You can replace R9 with adjustable resistor. P.S.Sorry for my written English, I am from Ukraine.

thank you for your answer, I would like to know if you have any idea how to reduce the range in this new RCWL-0516, the IC is RCWL-0001 which I do not find any datasheet.
[image]

Domeenick, 2024-01-27:
I have a pet water fountain with Rayde rdw037 motion sensors. They are too sensitive. Therefore, I connected the RCWL 0516 boards, but the device does not accept them. Sensors do not turn on. Perhaps the voltage from the device to the sensors is too low? Please help, photos below.
[image]
[image]
[image]

dDenVil, 2024-03-14:
So guys, what have you came up with? HOW to reduce range of the sensor?  R9 - is bad idea, also changing R11 or R12 - bad. So how to make it stable and reduce range?

stefan-sherwood, 2024-12-01:
Has anyone had any success reducing the range by blocking the sensor with material that reduces the range? As far as I've seen, the sensitivity is unaffected by walls and even metal objects.

Joerg-rw, 2024-12-01:
> Has anyone had any success reducing the range by blocking the sensor with material that reduces the range? As far as I've seen, the sensitivity is unaffected by walls and even metal objects.

see https://github.com/jdesbonnet/RCWL-0516/issues/22#issuecomment-676614390 in this very thread

Joerg-rw, 2024-12-02:
you can use a aluminum wallpaper to reliably limt the range to objects moving before that wall, nothing behind such large area completely conducting surface will get detected. I posted in another post in here how I made a directional sensor by putting the PCB in a carton box wich I plated on 5 surfaces with alu foil. That's narrowing down the sensing angle, it's not reducing the range, the latter is almost impossible.