jdesbonnet/RCWL-0516 #24: Change from Re-trigerable to NON-retriggerable

imayoda, 2018-04-04:
Hi there and many thanks for the research put on this nifty radar module.
I'm trying to put this module in non-retriggerable output, as mcu's interrupts won't be triggered again if the movement continues.. Output pin will stay always high if default re-triggerable mode is active and circa-continuous movement is detected..
I analyzed pin 1 (supposed mode selector) but it seems to connect 3.3V internally regulated voltage to the BJT's  collector, de facto putting this pin to ground disables the entire chip (short to ground).

I have also used the CDS pin to disable the chip, but in disabled mode it won't put the output pin LOW.

Does anybody have a suggestion on how to change detection behaviour ?
Kind regards

barewires, 2018-04-04:
The RCWL-0516 differs from the similar BISS0001 chip as pin 1 is now a 3V3 output from the internal voltage regulator. Output will be high on any motion with a minimum 2 second approx pulse width.

imayoda, 2018-04-04:
OK.. 
removed C2 and used the C-TM pads beneath to do some test.. almost non-retriggerable behavior achieved.
with 560picoF ceramic transistor all seems good, interrupt is firing many times per second, so I can decide how many violations have to happen in order to alarm the system :)
Easy mod:
[image]

barewires, 2018-04-04:
That is really cool, I too would prefer almost real-time signals and let it be processed externally. 
I put two on a PiZeroW in my window as a (neighbour's) cat detector as I wanted to AND the outputs. Node-RED is really quite interesting but there is no easy way to do Boolean, I would have thought that it would have been rudimentary. Still the detectors are too sensitive and watch me at my desk more than the cat.

How about a thread on wave guides and focusing?

https://photos.google.com/photo/AF1QipObwGu75XMrK1-HNRbKO80cfm8XtFh218MtOJUk

imayoda, 2018-04-04:
you can shield the sensor in order to achieve 180° detection (cats yay!), I use an aluminium foil connected to ground just behind the sensor.. cuts signal a lot
(can't see your link)

Test-subj, 2018-09-06:
Removing C2 and not adding anything on C-TM seems to trigger for ~6ms every ~300ms.

od1, 2019-02-12:
> Removing C2 and not adding anything on C-TM seems to trigger for ~6ms every ~300ms.
I tried that and it seems true, but then the trigger signal is around 0.1V.

Joerg-rw, 2019-04-07:
> The RCWL-0516 differs from the similar BISS0001 chip as pin 1 is now a 3V3 output from the internal voltage regulator. Output will be high on any motion with a minimum 2 second approx pulse width.

Nope, I checked by disconnecting pin1 from PCB and the operation changed from retriggerable to oneshot, just like you'd expect if pin1 is still identical function as in BISS0001 chip.
This means 
a) you can change to oneshot mode by simply cutting pin1
b) pjn1 is NOT a 3V3 output from the internal voltage regulator

barewires, 2019-04-13:
OK I see now that pin 1 is being tied to 3.3 v with VDD output on pin 11. I have loaded it to 100 mA and confirmed it works with VIN at 28 v although runs hot. I agree that the Pin 1 mode? description needs work.

dbuezas, 2019-11-16:
I'm in doubt. I want to get short pulses whenever there is movement instead of a long 3 second pulse.
Replacing C2 with a smaller cap like @imayoda  proposes worked (pulses are 200ms or so now), would disconnecting pin 1 (@Joerg-rw ) also does the same?
If so, how long are the pulses when pin1 is disconnected? 

Thanks!
(I'm not sure about what re-trigerable refers to. I assume it is about slope interrupts in a microcontroller not getting re trigered because the output is kept high (?)

Joerg-rw, 2019-11-17:
> I'm in doubt. I want to get short pulses whenever there is movement instead of a long 3 second pulse.
> Replacing C2 with a smaller cap like @imayoda proposes worked (pulses are 200ms or so now), would disconnecting pin 1 (@Joerg-rw ) also do the same?
> If so, how long are the pulses when pin1 is disconnected?
> 
> Thanks!
> (I'm not sure about what re-trigerable refers to. I assume it is about slope interrupts in a microcontroller not getting re trigered because the output is kept high (?)

let's assume a 3s pulse duration. "Retriggerable" means that any "movement detected" event  *during* that 3s period extends the pulse duration to end 3s after the event. So if a movement happens every 2.5s, with "retriggaerable=true" the pulse would stay active indefinitely, while with "retriggerable=false" the first event at T0 will trigger a pulse ending at T+3s, the second event at T+2.5s would have no effect whatsoever, pulse will end at T+3s and a new pulse would start at event at T+(2*2.5)s.

If you want a pulse to cause an interrupt on a GPIO of your rasPi or Arduino or whatever, your approach is to
   retriggerable=false
   Pulselength: very short. Use C2 to adjust pulse duration to 200ms, or even shorter, for the MCU a 1ms is ample time to detect the IRQ

dbuezas, 2019-11-17:
Thanks for the comprehensive answer!

> * retriggerable=false (not mandatory, might even be smarter to have retriggerable=true as long as your pulse duration is really short)

Why smarter?

Joerg-rw, 2019-11-18:
you use retriggerable=false if you want a defined pulse duration no matter what. Like somebody enters bathroom, doesn't move for 30s and then leaves bathroom. You set duration to maybe 60s and no retrigger.

 you use retrigger=true if you want to determine the duration of a movement event which is defined to end when for a certain timespan no new movement signal is seen. Like detecting walking, where there's a 0.05s of no movement between steps (ignore the lack of correctness in this example). So you set pulse duration to 0.10s and retrigger=true to detect end of arbitrary duration of a walk event. With retrigger=false you'd get a 0.1s pulse for roughly every second step, for retrigger=true you get one consecutive pulse until the whole walking action stops for longer than 0.1s.

I hope I was able to convey the idea behind the concept

Eszartek, 2020-09-27:
I've been trying to increase the pulses/sec output so I can handle all the sensitivity processing in my attached esp8266. 

I lifted pin 1 of the RCWL-9196 and removed C2, resulting in nice short 6ms pulses as noted by Test-subj. However, the 300ms lockout timer still limited the maximum pulses/sec to around 3.

Looking at the docs, it appeared that C3 controls the lockout time, so I removed it. Now I can capture 0 to ~26 pulses/sec and handle trigger set points in my code.

Right now I just look for values greater than 3(pps) and then trigger the alarm, as I get occasion 1 to 3 pps as noise, probably from the esp8266 just 4cm away. Now I will remote the sensor to see if I can eliminate the noise altogether.

imayoda, 2020-09-28:
enforce, with excellent capacitor, the power pins on the rcw nearest position, it helps a lot..
however I noticed that in RF noisy environments it gets really messy to use this radar module :(