jdesbonnet/RCWL-0516 #28: Pin 1 investigated

Joerg-rw, 2019-04-03:
I cut loose pin1 from PCB, powered up the module (worked unchanged!) and probed pin1:
1.4mV against GND - so definitely no output.
Obviously no effect on retrigger either :-/

/jOERG

CobraCalle, 2022-01-24:
Hello,

I`m totally confused at the moment... can anyone tell me the correct way to power this thing?

-3.3V On pin 1

or

->4V on pin "VIN"?

At the moment I have connected 3.3V (from rpi) to Pin 1 and it looks like it work like it should... but I`ve read that there is a LED on the board... but I do not see any powered LED. Now I have concerns that I do not use it the right way and that probably the detection range etc. does not work like it should in my case...

thank you very much

carl

Joerg-rw, 2022-01-24:
> -3.3V On pin 1
> or
> ->4V on pin "VIN"?
both should work, I'm not sure if #2 4 V are sufficient, might need 4.5 or 5V
#1 works but is "undocumented" and so doublecheck if any unexpected effects, particularly power consumption >>3mA

>I`ve read that there is a LED on the board... but I do not see any powered LED. Now I have concerns that I do not use it the right way and that probably the detection range etc. does not work like it should in my case...
> 
> thank you very much
> 
> carl

closed 2022-01-24 by Joerg-rw