jdesbonnet/RCWL-0516 #30: Help keeping main page updated

jdesbonnet, 2019-05-14:
Sorry folks, I haven't had time to keep the main page updated lately. If anyone wants to help with that I'd be delighted to add as project collaborator.

Joerg-rw, 2021-02-28:
feel free to add me, a tiny bit of my time may be available to dedicate to this task
/j

Cz-P, 2021-05-03:
Hi, The main description page needs to be updated in many places, like direct link to newer version (issue #32), BISS0001 and RCWL-9196 pins table, HF passive and LF active filters, Doppler effect formula (light and not sound speed!), 1MHz frequency change conclusions, Colpitts generator simulation, todo's etc. etc. 
Many of the items were discussed and clarified by number of posts.

wwalker, 2022-01-13:
Count me in too.  I'm new to the RCWL-0516, but I'll make the time.