jdesbonnet/RCWL-0516 #32: New/Different RCWL-0516 with slightly different layout

rradar, 2019-05-29:
Hello folks,

with my newest order I received six rcwl-0516 with a sightly different layout.

[image]

[image]

Most significant is the back with this new chip labeled '7133-1' which looks to be a 

> HT7133-1 30mA Low Power LDO The HT71XX-1 series is a set of three-terminal low power high voltage regulators implemented in CMOS technology.
> Source: https://www.digchip.com/datasheets/parts/datasheet/196/HT7133-1.php

Also the logos missing from the board itself as well on the rcwl-9196 microchip.

Some speculations what the added low power ldo could be good for?

MrRadiotron, 2019-05-30:
Shahriar over at The Signal Path shows how power supply voltage affects CW frequency for Colpitts oscillators on a similar module, the LDO would improve this significantly.

http://thesignalpath.com/blogs/2018/08/12/tutorial-experiment-teardown-of-a-24ghz-doppler-radar-module/

underwoodblog, 2019-05-30:
@rradar can you check if the HT7133 output goes to VDD (Pin11) and/or 3v3 output header? It looks like VIN from the header goes still elsewhere after the HT7133.  And if the internal LDO is still used? I can't see it on your pictures. 

HT7133 Pinout:
http://www.dzsc.com/icstock/uploadfile/200911210551209.jpg

Where have you bought the new version?

rradar, 2019-05-30:
@MrRadiotron thank's for your post which sound chinese for me :stuck_out_tongue_winking_eye: but improvement sounds always good to me :smile:

@underwoodblog I bought it over at these chinese store, let me check... aliexpress is the name! It's this item here: https://www.aliexpress.com/item/32879167410.html (even though the picture doesn't match with the one I got). The output on the front side is unclear to me. It sits under the rcwl9196 chip if I'm right. I don't have the skills do de-solder so can't really help with this in detail. I will try to make better pictures in day light maybe that can help...

underwoodblog, 2019-05-30:
Maby you don't need to desolder the chip if you have a Ohmmeter.  Meassure if the V-Out pin of the HT7133 is connected to PIN11 of the rcwl9169. These Schematics may help:

https://raw.githubusercontent.com/jdesbonnet/RCWL-0516/master/doc/john_taylor_rcwl-0516-0.png

It will take a few weeks til my new version arrives.

rradar, 2019-06-04:
Couldn't make any better photos of the board, sorry.

@underwoodblog my Ohmmeter is there but one probe broke and I need to wait my new one arrives. Let's see who's first :1st_place_medal: :grin:

rradar, 2019-07-18:
@underwoodblog finally I received my new probes... just seconds before I left my spot for a travel. So I can't give any updates for the next 3 weeks. Did you by accident already got your new version of the rcwl-0516?

underwoodblog, 2019-07-18:
no. hasnt' arrived yet. still waiting for over 20 orders from aliexpress. I think our custom office is in vacation.

ajinkya-open, 2019-08-07:
I tried this version of sensor it giving more stable and consistent output than non regulator version

even i am able to drive moc3021 through 330 by signal pin of sensor

comments are welcome

Nemesis7, 2020-08-16:
> I tried this version of sensor it giving more stable and consistent output than non regulator version
> 
> even i am able to drive moc3021 through 330 by signal pin of sensor
> 
> comments are welcome

I can confirm that this version works consistently over time, better than the previous version. My challenge now is to reduce the range using the R-GN points...

oldstu, 2021-01-14:
I'm new to github and interested in this radar board.  I have the version with the HT7133-1 and can confirm, using a multimeter, that its OUT pin goes to pin 11 of the RCWL9169 chip, that its IN goes to VIN of the 5-pin header, and that its GND goes to the GND pin on the header, as anticipated by earlier contributors.

For me, the next thing to try is dropping C2 on pin 4 by a factor of say 10 to 1nF to get a short trigger time, and increase R11 somewhat to drop the transistor bias and thus the power I hope.  Someone will have tried this already I expect.

Really, it is not a practical replacement for PIR sensors in battery operated circuits since a continuous 3mA drain is way too much, but an interesting device anyway.

Cz-P, 2021-05-03:
Hi, There's something wrong with depicted pin assignment - let's compare with proper schematic diagram.
Should be no problem with RC timers tuning to adjust Tx and Ti to your needs (requires 0603 1608M parts).
Playing with microwave stage is risky without well equipped lab. Changes in bias have impact on junction capacitances, feedback rate, generation stability, frequency hopping and radiated power, resulting detector sensitivity etc. The 2-3mA is probably about min, unfortunately.

Cz-P, 2021-05-03:
Hello, Would like to share some issues regarding the **newer version of RCWL-0516** sensor with external LDO. Its name hasn't change and most sellers haven't even realize it is a new part. Probably the only "updated" datasheet with **actual schematic diagram**, still without exhaustive and error-free description can be found at: 
 https://www.mantech.co.za/datasheets/products/RCWL-0516.pdf 
Please note strange REWL-0516 marking visible only on the one picture and mistakenly named pins (pin16=IOUT, pin15=M1N) on schematic. 
Other valuable sources include **EIRP** measurements in anechoic lab chamber, including **3D emission pattern**: 
 https://antennatestlab.com/antenna-examples/radar-antenna-pattern-rcwl-0516-hb100-cdm324 
 https://antennatestlab.com/wp-content/uploads/2019/07/RCWL-0516.mp4 
Measured and calculated total radiated power TRP was -15.9dBm 26μW.
Collected **parameter charts** for multiple values' additional elements can be found here: 
 https://bogza.ro/index.php/Microwave_Radar_Detector_Module_RCWL-0516#R-CDS 
Some better description (in Polish) is hidden on vendor page: 
 https://abc-rc.pl/product-pol-9463-Mikrofalowy-sensor-ruchu-4-28V-RCWL-0516-Czujnik-Arduino.html?query_id=1#showDescription 
Note: shop item parameters and pictures are not good (still unchanged old version), the text is wrapped under FAQ "Najczęstsze pytania ... **  v  ** " at bottom of the page.

Several articles were published few years ago when such modules entered the market - pictures, descriptions, schematic diagrams, simple circuits, microcontroller scripts, short films showing preliminary tests, usage ideas, and even radiation patterns as well as some problems, advices, uncertain theories and question marks. But unfortunately it seems that new entries ceased to appear, to do lists had not been completed and many suppositions were neither confirmed nor corrected...

Module specifications often omit emission frequency range of 2.9-3.2GHz, allocated for radionavigation and radiolocation, as well as compliance with local regulations.

Current draw is 2-3mA (no load), radiated power is only 20-30μW (short antenna). Due to small dimensions compared with 9.5-10cm wavelength radiation and detection pattern is not directional (yet not uniform), with the worst sensitivity in terminations end. The "lettuce" characteristics may have importance in open spaces, less inside buildings.

The high variation in transistor average current may be probably allocated mainly to generation changes, and only partially to npn parameters scatter and temperature.

There's a note that MMBR941M 8GHz transistor can be used instead of discontinued BFR520 9GHz [32W], however [N2] marking code suggests installation of BFS520 9GHz (sf up to 2GHz). They are both very similar, yet designed for higher collector current.
The R7 resistor seems to be a PIR circuit relict.

Dual opamp amplifier stage has max gain at about 3Hz with pass band of roughly 0.5-15Hz. Module normally does not react neither on very slow nor fast changes/moves.

Allowable OUT CMOS output load, with serial limiting 100R resistor, is less than +/-10mA source/sink. Can be referenced both to GND as well as 3V3. Good for driving of small transistor, LEDs or optocoupler.

New modules include control chips with or without markings (though voltage controller and HF transistor have always marks), it's hard to judge whether they are regular but proprietary or mutated ICs. Its "compatibility" is only partially true.
Adoption of external weak LDO was probably imposed, besides further production cheapening, by resignation from supply of higher current circuits from common line, as asynchronous disturbances can easily trigger false pulses, as well as temperature limitation, as chip and module could get too hot.
It means that soldered "special" control ICs have probably no or inoperative LDO inside, otherwise it would have no sense - the built-in stabilizer was more than sufficient for 3-30+mA.

New and old sensors are somehow unpredictable - they are susceptible to arrangement of wires and conducting parts, noise from nearby circuits as well as temperature. All strong echoes from reflections, interference from WiFi devices, computers and similar sensors may disturb or mask useful signals, triggering wrong pulses or making sensitivity worse.
It is also reported that detection sensitivity decreases in time, but the root cause was not yet explained - humidity, parts aging or other factors (#40).

Their parameters vary a lot and modules require individual tuning. PCB is provided with pads for i.a. additional R-GN/R3 resistor soldering for gain reduction/range shortening in some applications, but it appears that more often replacement of R6 or/and R5 22k for lower value (or R4 or/and R17 1M for higher value) is necessary to significantly increase the gain just to maintain some detection radius.

Cz-P, 2021-05-03:
### RCWL-9196 IC
Hello, I think the chip may deserve separate entry/thread, however some assumptions can be found elsewhere.

It is probably impossible to find its datasheet. Lack of vital data is completely unprofessional. The only info is that it is "compatible" with popular PIR controller **BISS0001** - except pin8.

Obviously there are more differences, like the built-in 3.3V 100mA **voltage controller**, resulting in much larger or dual die and higher temperatures due to heating.
Pin8 Vin is 4-28V power supply _input_ to the voltage controller. High-resistance reference voltage Vrf & \Reset inputs are most likely internally connected to the regulator output and Vdd (pin11).
Pin11 Vdd _input_ has additional function of regulated 3.3V supply _output_ for external circuits.
All other pins and units have equal functions, despite of slightly different marking on schematic diagrams, pin1 Mode=A _input_, pin9 VC=Enable _input_, pin2 VO=OUT _output_ etc.
The chip on new RCWL-0516 module drawing has "mixed" designations (pin8 VRF VIN, pin11 VDD 3V3), and PCB connections make it possible to utilize either IC - regular or "special" RCWL-9196 as well as BISS0001.

Specification of allowable junction temperature and thermal impedance or max power is maybe unnecessary for extremely low power ICs, but is indispensable for power chips containing LDO. Nevertheless unknown power dissipation cannot be high due to hybrid part and low thermal efficiency package, most likely less than 400-500mW.

Full current is probably only possible for couple volts voltage drop. Outside a range of 5~8V **derating** must be considered, with max current of maybe only 10-20mA at 28V depending on temperature and cooling conditions.

Performance and characteristics of the voltage controller are also unspecified. Supply of circuits with unsteady currents is risky as any disturbance on 3.3V line can cause false output pulses - such circuits should be supplied from separate controllers.

Integrated voltage controller can be omitted (pin8 hanging or rather shorted with pin11) and replaced by external stabilized supply, like implemented in new version of **RCWL-0516** with **HT7133-1** (in fact it is sometimes not certain what IC is used - let's say 9196-like clone).

Load of **VO** output is probably only +/-10mA source/sink (or less for 3.3V). CMOS levels of L 0V and H 3.3V are given for no load. OK - nothing is specified as there is no official data, yet I mean some parameters given for modules, which originate from the IC.

There is one more thing to analyze - change of pin9 **VC** input from high to low state during Tx period has no immediate effect and detector logic still reacts on VS signal resulting in output pulse elongation (in default retriggering Mode=1). 
Such behavior is required (and observed!) for control of switching of the light by the module equipped with daylight CDS (without additional logic or screens) - module must not react on light during output pulse. VC can block VS only after Tx+Ti. 
Otherwise (in case low VC would block VS) light would be switched on by move detector only for Tx, without possibility of Tx++ elongation/retriggering/recounting by following moves.
And this is clearly different from BISS0001 signal runs provided in known datasheets.

Another small difference are reference voltages (internal resistive ladder) - for BISS0001 VM=0.5*Vrf=0.5*Vdd VH≈0.7*Vdd VL=0.3*Vdd (symmetrical) thus VM≈1.65V @ 3.3V, whereas for RCWL-9196 VM≈2V @ 3.3V - so VM≈0.6*Vdd (shifted-up).

Tx and Ti times depend on RC values but also on Vdd voltage. It may be noted that provided formulae differ from maker to maker and are also inconsistent with tabulated values.
Direct formulae for approximate calculation of Tx and Ti based on resistances and capacitances connected to generators/timers respective pins are most often Tx≈24576*R2*(C2+C1) Ti≈24*R13*C3 or sometimes 2x longer Tx=49152*RR1*RC1 Ti=48*RR2*RC2 (where RR1=R2 RC1=C2||C1 C1=C_TM RR2=R13 RC2=C3 in case of RCWL-0516).
Formula utilizing measured R1C1 generator/timer frequency f1 is given as Tx=32768/f1. Similar formula for R2C2 and f2 is not provided - probably Ti=32/f2. But it may be less intrusive and more simple to measure time.

Parameters and PIR application schematic for BISS0001 are given for Vdd=5V whereas RCWL-9196 is typically supplied from internal or external Vdd=3.3V.

There are several similar variants of BISS0001, including some of wider supply voltage range, and some completely different (more complex). There's also a version in reduced 8-pin package (used in RCWL-0515 module).

All the above is based on some experience of new version of RCWL-0516 with U2 LDO and secret U1 IC as well as available internet materials. Appreciate confirmation with other observations.