jdesbonnet/RCWL-0516 #34: RCWL 0516 OUT Pin to Drive a MOSFET

sanjeevkolelar, 2019-06-23:
I want to switch ON a MOSFET when the OUT Pin of RCWL 0516 goes High and switch off MOSFET when it's Low.

I connected the OUT pin directly to the gate of a MOSFET, but the MOSFET is not completely ON. 

My Input voltage to RCWL 0516 is 6.8V, 4 mA.

Connected Load is LED.

MOSFET used is 5N50 ( 500V, 5 A). TO 220 package.

Please suggest.

pokergeist, 2019-08-03:
The output voltage of OUT is 3.3V when on. That's too low to turn on your 5N50 MOSFET (~6VDC). You need a low voltage logic level device.

I used an N-/P-channel Si4559ADY with the n-channel MOSFET gate driven by OUT. You could just use the n-channel MOSFET as a low side switch, but I used it to turn on the p-channel MOSFET for 12VDC high-side switching for an LCD strip (~ 1.6A). 

I also used BSS806NE n-channel MOSFETs driven by a 3.3V ItsyBitsy microcontroller for low-side switching of an RGB LED strip. The OUT signal is used to retrigger the uC for task lighting mode for kitchen counter lighting mounted under the cabinets.

I made boards for both applications. The single channel board mounts under the RCWL board and has two 2.1mm barrel connectors underneath for 12V and the LED strip.

Joerg-rw, 2021-02-28:
you need a logic level mosfet - i.e. one that has a low (<<3.3V) Vgs_on, like several of the IRL* series.
See IRL540 for example, refer to https://arduinodiy.wordpress.com/2012/05/02/using-mosfets-with-ttl-levels/

tsaost, 2021-06-12:
I use the n-channel AO3400A which is very cheap (less than $2 for 100 on aliexpress).