jdesbonnet/RCWL-0516 #35: RCWL-0516s close to zero sensitivity/triggers outdoors, work great inside at the bench.

dsbsu, 2019-07-03:
Hello,
I am having a very elusive problem with all five RCWL-5016 modules that I recently purchased not working when moved outdoors.  Inside my garage, with the overhead door open (same temperature and humidity) the sensors work great, approximately 15 to 20 feet detection area in most of a 360 degree coverage area.  When I take the same sensors outside, even on a breadboard with no other circuity, I get a reduction of almost 98% in sensitivity.  I can move my hand one inch away from the board and it will light up an attached LED.  My guess was RF interference, so I killed every breaker in my house (except the known working AC from my garage outlets) ran an extension cord out to my yard from the garage, and still had the same problem.  So, no RF from my house, and the next neighbor is at least 80 feet away.
I noticed at the work bench that Pin 12 of the main IC was the most active by motion near the board, almost a 3 volt swing with activity within inches or several feet away from the board.  Again, out in the yard I could only duplicate such a swing in voltage on Pin 12 if my hand was within a couple of inches from the RCVL-0516.
Any thoughts?  Appreciate your input.
Dennis

geologic, 2019-07-31:
Any news on this issue? I have the same problem, indoors it works ok detecting movement at 5m; going outdoors into open space, sometimes detects me walking towards just when i'm some centimeters near.
My setup is very simple, batteries and a LDO @ 5V to power module, and a LED on output to check detection.

bkscharf, 2019-08-01:
Just a thought... interference from nearby cell towers?

dsbsu, 2019-08-01:
I have not found any solutions to date.  It must be some type of RF 
interference, but I experienced the same problem when doing a teat about 
three miles away from my home.

dsbsu, 2019-08-01:
Cell tower interference has crossed my mind.  I use AT&T for mobile, and 
the signal at my home is very weak.  Not sure about other carriers?  I 
plan to keep a test setup in my car, and try it at different locations 
around the state as I travel.

Anyone know of a RF spectrum analyzer app for a mobile device?

Thanks...

geologic, 2019-08-01:
I doubt could be RF interference... I tested on the countryside, where GSM is very weak, same behavior.

Could be lack of multipath? inside my house the walls/metal should reflect back some microwaves, in open field reflection should be less...

mkarliner, 2019-10-06:
Just done a test with one of mine, and it suggests that the multipath explanation is the correct one.
In the house with a ~10M hallway, I can get a response all the way down the hall.
In the garden, I get maybe 3M maximum response. Now, my garden is suburban, with a shed and fences nearby, which might explain the better performance than the original poster's.

jeffreyclay, 2020-01-02:
I would guess the outdoors limits a strong return echo for the chip to process. The out going ping simply dies by normal propagation (inverse square law)

dsbsu, 2020-01-03:
Your assumption is most likely correct.  I moved one of my units to the 
south side of my house that has a wrought iron fence along the border. 
That unit seems to work and is probably using the fence to reflect the wave.