jdesbonnet/RCWL-0516 #36: Repeat Trigger time?

dezkok, 2019-07-05:
I've been through the notes and as far as I can see, this question hasn't been covered. The "on" time where the output pin is held high is around 2 seconds, I see from other notes that this can be increased by putting a capacitor across C-TM, but I want to significantly DECREASE the time the output is held high and trigger time. I've tried various resistors across C-TM and can increase the time, but have had no luck in decreasing the time. Perhaps explaining what I am hoping to achieve would help. I'm trying to create a pulse count circuit where, the system will only trigger after two or three pulses within 2 seconds (similar to alarm PIR detectors) to reduce  false alarms. Has anyone been able to decrease the trigger time?

daisie-security, 2019-07-25:
You can see by looking at the schematic (https://raw.githubusercontent.com/jdesbonnet/RCWL-0516/master/doc/john_taylor_rcwl-0516-0.png), that to decrease the on time, you can do one of two things; either remove C2, and replace with a SMALLER capacitor than C2 across C-TM, OR add an additional resistor across R2 (between IC pins 3&4). The on time delay works by charging C2 using R2, which is why placing an additional cap on C-TM can only increase it because C-TM & C2 act additively in parallel, whereas placing another resistor across R2 lowers the effective charging resistance and therefore increases the current flow and rate of charge of C2.

johnwigley, 2019-07-25:
You can see by looking at the schematic (https://raw.githubusercontent.com/jdesbonnet/RCWL-0516/master/doc/john_taylor_rcwl-0516-0.png), that to decrease the on time, you can do one of two things; either remove C2, and replace with a SMALLER capacitor than C2 across C-TM, OR add an additional resistor across R2 (between IC pins 3&4). The on time delay works by charging C2 using R2, which is why placing an additional cap on C-TM can only increase it because C-TM & C2 act additively in parallel, whereas placing another resistor across R2 lowers the effective charging resistance and therefore increases the current flow and rate of charge of C2.

I wanted to do the same thing, and ended up finding the easiest thing to achieve soldering wise was to desolder C2, and then put a 1nF cap across C-TM, so you get something like 200ms trigger pulses out of it.

dezkok, 2019-07-25:
Thank  you. Much appreciated. I'll give that a try and let you know how it works out. Thanks!

Nestcomputers, 2020-08-30:
Hi please can you giving me value of capacitor for C-TM to increase delay triggered time for 1 min. As I want to increase delay trigger time to 2 min to 4 in different application.

SUSTHESURFER, 2020-12-21:
But inserting 100nano farad capacitor at C-TM you can increase the delay time to 40 seconds. For more details visit [Automotionsensor](https://automotionsensor.com/rcwl-0516-microwave-radar-sensor/) .

dlin4668, 2021-03-03:
@SUSTHESURFER: Please correct if my formula is incorrect:  2s * (100nf + 10nf)/10nf = 22s, not 40s? Thank you!

dlin4668, 2021-03-20:
After removing C2 and installing C1/C-TM with 220pf, I can get a much narrower high pulse as predicated with this formula: 2s * (0.22nf)/10nf = 44ms,

However, no matter how I move and stop, the smallest cycle time is at least 2s. Is there any method to reduce this cycle time (i.e. the frequency of OUT pin is < 0.5Hz)? See the waveform below

[image]

cynosaurous, 2021-05-02:
I had the same problem:   A single hiccup, and I'd get a 2-second pulse at the output.  False alarms about once per day.  I figured that if I could get very short pulses, and count several of them within a window, I could be sure there was really something moving.
Do get this, I had to do two things:
1) I replaced C18 with a 0.001 uF capacitor.  (See the schematic/placement diagram mentioned in a comment above.)
2) I lifted pin 1 of the IC, and wired it to ground.   This changes the mode to non-retriggerable, so that the output pulse is not kept high by successive triggers.

With the C18 I chose (actually, that's what I happened to have  available), I get an output pulse from the board of 307 - 308 ms.    If I wave my hand, I get a series of 308ms pulses, separated by varying amounts.    The smallest gap between pulses I've seen is about 240ms.    If I move my hand slowly, I'll see 308ms pulses with separations of 1 second or more.    To detect motion, I look for 3 or more pulses in a 5 second window.   Few false alarms (except when my neighbor's car pulls out at 5 AM).

LorenKonkus, 2022-07-25:
In the same way that you can replace C2 with a smaller capacitor for a narrower pulse, you can replace C3 with a smaller capacitor for a shorter retrigger time.  Replacing both with .001 uf results in a minimum output pulse duration of about 300 ms and a minimum time between pulses of about 300 ms.