jdesbonnet/RCWL-0516 #37: Ditch the IC, use just analog circuitry connected to uC ?

Harvie, 2019-08-31:
I am designing my own board with microcontroller (either ESP32 or ATmega), i would like to include the doppler sensor design directly to that board. I am bit tight on space and i was thinking that maybe i can just use the RF part of this circuit and replace RCWL 9196 block with my own circuitry to detect the output frequency shift myself. Effectively i want to cut the RCWL-0516 part in half and throw the IC half away.

I guess we can detect the baseband (downconverted doppler shift signal) using ADC of microcontroller. Maybe with little help of LM358 opamp or something...

Any ideas how to do this? Unfortunately i don't have scope at hand :/

Harvie, 2019-09-11:
[image]

[image]

Harvie, 2019-12-29:
OK, so we've managed to design the radar to custom microcontroller PCB while replacing RCWL-9196 IC by commonly available opamp. I will post more details later...

chaotix-, 2020-03-08:
That is very interesting. Did you get it to work? Do you have more information on this?

Harvie, 2020-03-08:
Yes. It is working including the analogue signal. But i am just very busy. Hopefully i will manage to make Kicad design and upload it, but i have no idea about ETA.

chaotix-, 2020-03-09:
Please do, when you finde the time and post a short note in this issue. Thanks
a lot.

paddyrg, 2020-05-13:
I'd also love to see your results, it could be quite handy for me in the next couple of months!

mimeakadug, 2020-10-19:
Hi all,
Please checkout my work here https://github.com/mimeakadug/Doppler_Radar_RCWL-0516. I have sampled the baseband signal using an arduino and processed the signal to generate a Fourier plot while the radar is looking at a ceiling fan. The Ceiling fan induces multiple Doppler velocities as seen the photos below
[image]
[image]
thanks
GD

mimeakadug, 2020-10-21:
[image]
-Added STFT algorithm for spectrogram display. Getting more meaningful results.
-Power spectral density i.s plotted too. Time domain signal converted to volts (V)

Hypothesis -> 18 peaks in 1.2 seconds -> 15 peaks in 1 second -> 5 revolution per second (divide by 3 fan blades) -> 900 rpm which is approximately how fast a ceiling fan rotates at full speed.

dbuezas, 2020-10-21:
Would a higher time resolution be useful for you? There are ways to speed that up quite a lot (e.g. changing the ADC clock prescaler, ditching analogRead, acquiring and sending in batches, ...).

mimeakadug, 2020-10-21:
@dbuezas  A higher time resolution will mean I can detect higher Doppler velocity shifts. I can already detect up to 37 m/s with the RCWL-0516 which would be alright for most things. 

I am planning to try this code with another Doppler radar which has a higher Carrier frequency hence would result in higher Doppler frequencies for the same velocity of the target so for that it would be helpful. I would appreciate any changes / suggestions you would want to make my code I see you are a much better coder than I am :)!

dbuezas, 2020-10-22:
These are really interesting experiments you're doing. I'll gladly help if you need a couple of tricks to speed up readings.
I recently spend some time doing exactly that in this project https://github.com/dbuezas/arduino-web-oscilloscope
(not for standard arduino boards yet)

Harvie, 2021-03-16:
I have some work in progress KiCAD design of ESP32 devkit with doppler radar daughterboard:

[image]

I've used bit different radar design, which does not need any shapes on the back side of PCB. Just ground plane. This is really great, because single sided boards are easier to make at home. (well actualy dual sided, but ground plane does not require alignment or any complex etched pattern on back). Also this makes it possible to use 4 layer board in scuh way that 2 layers are used for radar and two more layers are still left for microcontroller and other stuff. I've unfolded this to separate daughterboard, because it is expensive to do pick and place on both sides at the same time, but it can be cut and folded back on itself.

Unfortunately there are several issues (both technical and legal) i need to solve before i will be able to share the files.

chaotix-, 2021-03-16:
> Unfortunately there are several issues (both technical and legal) i need to solve before i will be able to share the files.

This sounds very promising. Please share when you have solved the problems. I'm still very much interested.

danilozte, 2022-01-26:
Chaotix como vas con los diseños? ya solucionaste los problemas tecnicos de diseño?, quiero hacer mi propio diseño.
Chaotix how are you doing with the designs? have you solved the technical design problems, I want to make my own design.

Harvie, 2022-01-27:
There was not much progress since in terms of design. The legal problems are mostly solved.
I think i will need to review the files to see if all chips are still available for PCBA, since global chip shortage messed us a bit. Then i will order the prototype batch to verify the design. Hopefully i will find some time to do this as i am still rather busy.

I know it's been long. But i had not abandoned the idea yet :-)

willtmakinen78, 2022-06-07:
Harvie, any updates in the last 4 months? Even if the design isn't fully validated I would love to see the KiCAD files and/or see a schematic, since I would like to do something very similar and your files could serve as a great starting point.

Harvie, 2022-06-08:
Yes, sorry. i am busy. i will try to share the design today even when not fully routed and tested. however we had some success with it in the past.

Harvie, 2022-06-08:
Ok, here are the files:

[REF-radar.tar.gz](https://github.com/jdesbonnet/RCWL-0516/files/8859532/REF-radar.tar.gz)

Some notes:

  * it is in Kicad 6 format, footprints and symbols are part of the schematic and pcb, but can be exported if needed
  * Radar footprint requires precise placement of other components over it in order to allow automated PCB assembly. ratsnests will indicate the corect place
  * It is not required, but you might want to add ground copper fill around the radar, it should fuse with  it without any gap.
  * Back side does not contain any components, it's just full copper fill, therefore you can use 4 layer board and have the backside as internal layer still allowing you to use remaining 2 layers for another purposes.

Eg. Here you can see ratsnest indicating where exactly to place the resistor for PCB assembly
[image]

willtmakinen78, 2022-06-08:
This is great, I'll have a look at it and let you know my thoughts and if I have any questions

willtmakinen78, 2022-06-08:
Harvie, I'm confused about the purpose of this IC in the schematic. None of the components attached to it appear in the PCB file. Additionally, what's the difference between the 3v3, Radar_3v3, and the VCC lines
[image]
?

Harvie, 2022-06-08:
> Harvie, I'm confused about the purpose of this IC in the schematic.

That is not IC. This is symbol which is associated with the radar RF footprint. Or you may call it "the pcb antenna". There is no actual component populated to this footprint. All components have separate footprint. This might look rather cryptic at the first sight (especially in the schematic), however it allows you to easily integrate this antenna footprint to your custom PCB without having to carefully craft the RF layout every time, which was my goal. Also it allows you to generate data for automatic assembly, because each component has it's own footprint overlayed on top of that antenna. i wouldn't able to figure better way of doing this so that design remains both easily reusable and fit for PCBA.

> what's the difference between the 3v3, Radar_3v3, and the VCC lines

There is transistor which uses digital signal to switch 3V3 that goes to radar, so you can disable it using microcontroller when not needed. You can ditch the transistor circuit and let it powered on all the time.

willtmakinen78, 2022-06-08:
That makes sense. Where in the PCB file is the BFR193 Q4? Only the  Si2301CDS Q3 is present on the current layout.

Harvie, 2022-06-09:
> That makes sense. Where in the PCB file is the BFR193 Q4? Only the Si2301CDS Q3 is present on the current layout.

You are right. you wanted it fast, so i haven't checked almost anything :-) The REF numbering do not match between schematic and pcb, you might want to fox that. But there are only two transistors. One optional which switches the whole thing on and of. Second is part of the RF design of radar. All of the parts which are only directly connected to the "antenna" symbol are placed on top of the RF footprint. Therefore the RF transistor is this one:

[image]

Harvie, 2022-06-09:
Just pres this button in PCB

[image]

Then select "delete footprints without symbols" and "replace footprints with those specified in schematic"

it will sync

willtmakinen78, 2022-06-09:
Another question, what is the inductance of the microstrip inductor (squiggly trace), and the parameters of the rectangular antenna traces? Also, what frequncy does it oscillate at?

Harvie, 2022-06-10:
> what is the inductance of the microstrip inductor (squiggly trace), and the parameters of the rectangular antenna traces?
 
TBH i don't know, it will vary on different boards due to thickness and material differences, etc... You should not use board which is too thick, i would stick with thinner ones if possible. I don't have equipment required to measure these, however you can find online calculators, put dimensions in and you will get some approximate calculation if that's important to you.

The design is based on some chinese module, but we've replaced baseband circuitry with opamp so we can process the output with ADC on ESP32. (original only had digital output which latched in active state for few seconds after each trigger). This modification was proven to work with microcontroler (even on rather thick board), but not proven to comply with regulations.

> Also, what frequency does it oscillate at?

I beleive it was little bit over 3GHz, but never really measured it, so take that with huge crystal of salt. Fortunately this circuit is not very sensitive to frequency variation. Like most of the cheap doppler radars, it uses same oscillator for TX and RX, so it can receive whatever frequency it transmits, so basicaly it subtracts the transmitted frequency from received signal and you are in turn left with just the difference between the two, which is rather low frequency which can be easily processed in the baseband amplifier and captured to uC using ADC. Exact RF frequency is not that critical (unless we speak about local regulations)

You might even try to use different PCB thickness to change the frequency until you get to some freely available part of specturm. Otherwise you might try to change dimensions of the RF traces a bit.

willtmakinen78, 2022-06-10:
Thanks. A final thing I was curious about was the choice in corner frequencies of the op amp bandpass filter - after simulation they seem to be around 100Hz and 100KHz. For person detection, a doppler shift of 1-100Hz would be more expected, as these frequencies correlate to movement speeds of 0.5m/s-10m/s. Obviously you gotworkable results from the current circuit, but I was wondering what your thought process was. Does the atennae and supporting circuitry shift the difference in frequency up in some way?

Harvie, 2022-06-10:
> what your thought process was

it was quite some time, but i beleive the original circuit used some application specific IC, which also had integrated amplifier along with other things, so we just took that circuit and replaced "weird" chinese IC with more common LM358 standalone opamp so we only get the unprocessed baseband signal.

> after simulation they seem to be around 100Hz and 100KHz.

interresting. what are you using for the simulation? never tried simulating anything RF.

> For person detection, a doppler shift of 1-100Hz would be more expected, as these frequencies correlate to movement speeds of 0.5m/s-10m/s.

Great. This is exactly the reason why i shared the design. So smart people like you can figure out new ways to improve it :-) Should be simple as replacing some RC filters right? You can order several PCBs for cheap and try populating different baseband filter setups and let us know what works best for you. Just note that between 1 and 100Hz that you propose lays huge ammount of noise caused by 50/60Hz mains and these devices are often mounted directly to light fixtures, lightbulbs, etc... Might that be an issue?

> Does the atennae and supporting circuitry shift the difference in frequency up in some way?

I have no way to measure this

willtmakinen78, 2022-06-10:
I just used LTspice for simuation, here's a screenshot of the bode plot, starting from th initial low pass filter consisting of R26, R27, R30, and C24-C27 to the output of the 2nd stage of the amplifier.

[image]

Defintely some room for playing around here - I was going to do exactly what you suggested and get a few boards made up with different gain and corner frequencies and see which ones work best, as well as using a scope to see what the doppler shits being produced by the antenna actually looks like. This should also let me see what sort of mains interference occurs as well.

willtmakinen78, 2022-06-10:
> The design is based on some chinese module, 

Any chance you know which module?

Harvie, 2022-06-10:
> Any chance you know which module?

We've been working on this with several people. I had this idea to add analog output to chinese module and i was helping with the design, remotely instructing other coworkers how to do it, while i was on vacation, and specificaly told them to use RCWL-0516 (because i've previously managed to modify the module for this purpose), but for some reason they've used some different design i've never saw before. But i liked the fact it is single sided (only groundplane on the backside, unline the weird circles that RCWL-0516 has on backside), so i was happy with this and didn't cared to enquire for more information. I can't really find what module was originaly used exactly, because it's few years ago. I can't even find photos of similar module online.

willtmakinen78, 2022-06-11:
No problem, I'd agree that the additional simplcity is a big plus. I'll get an antenna fabbed up using your kicad design (after converting it to altium, my preferred pcb design software) and see how things go.

Harvie, 2022-08-09:
https://www.youtube.com/watch?v=tFxKUzEDSdw

paddyrg, 2022-08-10:
> Radar using ESP32 wifi peripheral https://www.youtube.com/watch?v=tFxKUzEDSdw

Interesting!  I looked at the link Espressif suggested https://www.espressif.com/en/company/newsroom/news in the comments for any news...nothing in the past year seems to mention esp-wifi-csi sadly, so we are still waiting :,-(

Cowh3adDK, 2022-12-13:
[image]
Hi after looking at this thread, and wanting to incorporate a "hidden" presence detector in a lightswitch i am working on, i started to try and reverse engineer the amplification part, and i am not having much success, my oscilloscope, can easily catch the voltage differences on the RCWL module itself, but when i try and build my own op-amp circuit nothing works, tried both the one provided by harvie, and 1 a little simpler. now im trying one based on the values of components from the RCWL schematic and saw that 2nd stage op-amp does not have a positive input, is what i have put in my image okay to use?

casesolved-co-uk, 2023-10-27:
It looks to me like the components in the blue box are actually meant to be connected to the - input of the 2nd op-amp, and the + input is supposed to be to GND.

See inverting band pass filter circuit here:
https://www.electronics-tutorials.ws/filter/filter_7.html

The blue box components bias the input voltage (by charging the capacitor) to keep it within the voltage range of the op-amp. The AC coupled signal doesn't have a DC bias otherwise.