jdesbonnet/RCWL-0516 #38: uni-direction detection

BD981, 2019-10-10:
Can it be used for detecting uni-directional moving? Walking forward, back and especially parallel to the wall for example? I need to find out this issue. Could you please help? Thanks in advance.

Harvie, 2019-10-22:
In theory it might be possible to detect if object moves towards the detector or away from it. That's something that should be possible using doppler effect used in this device. However i can't really tell how complex such mod would be.

This video might give you some insight: https://www.youtube.com/watch?v=Kzsh59TM4MY

paddyrg, 2020-05-13:
Looking at the waveforms generated, it's way too marginal to do reliably.  And it's not true Doppler, either, so I'd suggest there are far better ways to get the same result

mimeakadug, 2020-10-20:
It is not possible to detect the direction of motion with this sensor since it gives you a real (in the mathematical sense) baseband signal at the output. We will need to make hardware modifications specifically we need an IQ demodulator so we get a complex (in the mathematical sense) baseband signal which on processing (Fourier Transform) can give you direction of motion. For more details you can checkout  "A Visual Introduction to DSP for SDR" here https://kb.ettus.com/Cyberspectrum