jdesbonnet/RCWL-0516 #39: Easy to kill by backfeeding voltage?

ligius-, 2020-02-10:
I had one of these connected to a rPi and it worked fine for about a month. Then I decided to hook it up to an Arduino and did not read the datasheet. I've fed 3.3V into the 3.3V and now it's seemingly dead. I did not put an oscilloscope to it to see if it actually does something, but the leds don't light up anymore and it doesn't draw any current.

I had a second module - hooked it up to an Arduino (just the 3.3V regulated out) and connected the output pin to a buzzer. It buzzed for less then a second and it went dead. Not sure if the buzzer killed it or the 3.3V.

Perhaps it would be useful to have a list of DON'Ts, such as: no voltage on the 3.3V output (without power), no inductive loads, no voltage on OUT... I don't really know what kills these modules. Static?

ligius-, 2020-02-12:
Well, it seems that at least the first module was not dead, just sleeping?! Moreover, it doesn't seem to be able to drive a piezo module, which is quite weird.
The piezo just has a driver transistor, grounding the base turns it on. I just shorted the base and ground (emitter) together. From the serial out of an Arduino it is still able to make some noise (pull-up) but somehow it's silent when driven from the microwave sensor. The sensor however can drive the low signal, so the piezo would sound only when there's no movement. Very strange, I will have to scope this.

LittleFox94, 2022-01-08:
has the piezo some frequency generation included to work with just a voltage being applied? if you only hear a "click" it might not?