jdesbonnet/RCWL-0516 #40: Decreasing sensitivity over time?

jonhendry, 2020-04-20:
Has anyone noticed a decrease in sensitivity over time? Like at first it senses motion around the corner in another room, and then two months later you have to be within six feet directly in front of it?

I have a motion-sensing non-PIR ceiling light that I installed in the laundry room which has exhibited this behavior. At first it could sense motion around a corner or through a closed, louvered folding door. Now you have to get pretty close.

I don't know which doppler module it's using, but I figured someone might have run into similar behavior with RCWL-0516-based units.

paddyrg, 2020-05-13:
I have strong suspicions of this, too.  One prototype device worked ideally in the lab and for the first month or two on site, but became increasingly marginal.

digamesystems, 2021-01-17:
Any updates on this? -- I've seen similar behavior in my lab. When I first hooked up the unit it worked great; c.a. 7 Meters of detection range for a person. Now it's down to 2-3 meters. 

Could I have ESD'd it somehow? -- I'd expect the only thing that would be sensitive to that might be the logic chip. 

I don't think it's an over voltage problem. -- I've been running it using the 5V output from an Arduino with a 47uF cap between 5V and GND.

rllaw1958, 2021-02-01:
I have been looking to use  this device in some of my projects and after listening to your complaints it seems to me the problem is probably environmental. This device relies heavily on circuit parasitics to function properly and with that in mind you must realize that FR4 is Hygroscopic and will absorb moisture over time. An increase in moisture to the PCB will change the dielectric properties of the FR4. (A quick google search will confirm) To test if this is your problem I suggest (if possible) to place your device in an oven at 250F for a few hours then retest to see if your sensitivity has been restored. The solution may then be a conformal coat or a hermetically sealed project box. 
Please note that too high a temp and/or too long exposure to higher temps may damage the PCB and discolor silkscreen. if you have any concerns the google search will net several documents on the issue.

XOIIO, 2021-04-24:
I hope someone tests the theory drying the board out, this would be good to know if it is an issue. Perhaps for long term use sealing it in epoxy would be necessary, if that doesn't affect it somehow.