jdesbonnet/RCWL-0516 #41: [How-To] Suppress the sensitivity at backside?

pranjal-joshi, 2020-07-06:
Hello,
I'm planning to use this sensor as a motion detector on my project.
The sensor will be mounted on the ceiling. But as I stay in an apartment, **What If someone walks on the floor above my ceiling, will it also create a false motion signal at the output?**
If yes, Can I suppress the sensitivity of the sensor backside by using a metal shielding, Ground plane of PCB or any other proven method?

uherting, 2020-07-10:
Recently I noticed that the sensor can get triggered by motion on the other side of a 12cm brick wall which has wooden beams integrated. 
I suppose it is because of the wooden beams, but I have to run further checks to confirm.

shr00mie, 2020-11-17:
I too would very much like to know if anyone's done any experimentation regarding materials and configurations used to possibly make the device more directional instead of what feels like a solid 360* pickup angle. I'm currently using it inside of a wooden box, and i've noticed that if the back is facing a protoboard, for example, in tight quarters, it'll constantly report motion. i'm going to try to incrementally increase the distance between the rear of the sensor and protoboard to see if there's some sort of threshold.

digamesystems, 2021-01-12:
For fun, I took mine, mounted it in on a solder-less breadboard and centered the the board in the middle of a metal bread pan with about an inch of space on all sides. It couldn't see through the metal but worked fine on the open side.

Bad ascii art below... 

"-" = pan walls
"*" = sensor

    ---------
    |
    |      *
    |
    ---------

digamesystems, 2021-01-12:
In a plastic project box, you might consider some copper tape. The specs recommend at least a centimeter gap between the board and any metal.

Joerg-rw, 2021-02-28:
https://github.com/jdesbonnet/RCWL-0516/issues/23#issuecomment-482768140

XOIIO, 2021-04-24:
> In a plastic project box, you might consider some copper tape. The specs recommend at least a centimeter gap between the board and any metal.

That's what I'm going to do, I'm looking to only detect motion at the front, and ideally in a somewhat narrow pattern beam, the CDM324 looked interesting, I ordered multiple sensors, but that one seems a lot harder to interface with an arduino.

I'm going to 3d print a holder that has that 10mm gap around the board on all sides and back it with copper tape connected to ground and hopefully that will work, and hopefully having some of the shielding stick past the front will help to shape the beam, I'll report back if I remember to after testing it out.

There is also emi absorbing foam that might be a better bet though that's more expensive.

I'm also looking to reduce the range, 5m is still pretty far, I'd like to get it down to 3 and hopefully a 2m resistor or something between 1 and 2 will bring the range down to that sort of distance.