jdesbonnet/RCWL-0516 #42: Reduced Range but increased Current consumption

deepaksisodia079, 2020-07-06:
I was able to modify the range to few cms 4-5 cms with 40 ohm resistance put in R9.I removed the original 221 ohm from its place.I also tried multiple resistance with R-gn but was not successful.

BUT

I am into a new problem,before any modification or keeping the resistance to around 220 ohm on the R9 resistor keeps the current consumption of the chip around 2.3 mAh but with the 43 Ohm resistance on R9 the consumption increased to 12mAh.

I am planning to keep the current consumption to be very low sine i want to run the prototype on a 4 x AA Alkaline battery which has a current of 1500 mAH give or take and with this rate the battery would die in few days.

Can someone help me with this problem,i am a beginner with electronics as i am more of a computer engineer?

Can I put something else along with the resistor to keep the range withing 4-7 cms and also draw less current?
Or
Is there any other way to reduce the power consumption of the chip by removing some pins from the IC on the board?
or 
Putting a resistor before the VoltageIn pin of the RCWL will have any impact?

Please help

dbuezas, 2020-07-07:
> Can I put something else along with the resistor to keep the range withing 4-7 cms and also draw less current?

Did you try increasing r8 instead of reducing r9? This should also attenuate the signal going in to the chip without messing up the resonance frequency of the antenna.

deepaksisodia079, 2020-07-07:
I will try today and share my results.
This is what I will do:

no change to R9 
only change the R8 to a higher value...

dbuezas, 2020-08-17:
Worked?

jepoy11, 2021-04-20:
Hi I tried to add change the R8 from 1k to 47k ohms, the detection range is now 3inch

Joerg-rw, 2022-01-13:
R3 (R-GN) is the one to add a potentiometer to. GN like "GaiN"
Reducing R8 to 10R detunes the filters, messes with opamp DC bias, and same time massively increases the signal strength. If it would result in a lower sensitivity, then only via the detuned filters or the opamp1 running into clipping. Strongly deprecated