jdesbonnet/RCWL-0516 #44: Electronic disturbance

Laxi-luke, 2020-09-26:
Hi,

This is actually a general question for radar/microwave detectors but the RCWL-0516 was the smallest one I could find with a quick search online.

Does anyone know if it can cause disturbance to other electronic? The questions is asked since I'm searching for a way to detect motion from inside an enclosement where other electronics is placed.

Grateful for any support!

mimeakadug, 2020-10-20:
Generally radars operate at higher frequencies and have separate bands reserved in the radio spectrum so they don't cause interference (eg automotive radars operate at 77 GHz). For the RCWL-0516 the frequency of operation is 3.101 GHz but I am not sure if it generates any other harmonics since there is not much data available on it online.  Additionally the sensor itself might be susceptible to interference in terms of mutual inductance - caused by having metal close to the sensor. If you could provide more details on what electronics you plan on using I could maybe do a few specific tests for you.

sa60t3ur, 2021-10-04:
Is there any way to detect speed of objects using `RCWL-0516` or `CDM324` or `HFS-DC06H`?
I had googled; I didn't success with all the related content/documentation online.
Is there any working example available? or detecting speed isn't `TRUE`.