jdesbonnet/RCWL-0516 #45: How to detect an object getting closer to RCWL-0516

Narcissus-GBC, 2021-02-27:
Hello,
 I'm doing a project and this is the first time I work with sensors and I have no experience.
  What is your suggestion to change the sound of a buzzer, when RCWL-0516 has detected an object? I mean, when the object is getting closer, the louder and louder sound could be heard from the buzzer.

JohaGit, 2021-03-07:
Already covered in another thread here: https://github.com/jdesbonnet/RCWL-0516/issues/11#issuecomment-792257689

duplicate topic. Suggest closing.

Joerg-rw, 2021-03-07:
This sensor doesn't detect distance.
It detects level of reflected signal multiplied by frequency deviation caused by Doppler effect, as well as variations in mere destructive interference when reflected signal is 0° (no cancellation) vs 180° (max cancellation) phase difference https://en.wikipedia.org/wiki/Phase_(waves) to original oscillator signal