jdesbonnet/RCWL-0516 #55: Helping my friend with a cheap Chinese radar light bulb

Halo3fanz, 2022-04-08:
Hi all,

My friend purchased a light bulb with a radar sensor. It lists for SGD14 (US$10) but is perpetually on sale for SGD4 (US$2.93):

[E27 Sensor Radar Light Bulb](https://shopee.sg/product/164768716/7115180590?smtt=0.192909559-1647008577.10)

However it's detection range was too large (~4m radius). I stumbled on this project and realised the circuit and principles used are similar. Here is the circuit:

[image]

[image]
  
And my messy amateur interpretation of it:

[image]

The chip is a [HYL-1617](https://www.taobao.com/list/item/641087556617.htm) (not an easy find and you need a Taobao account) which can be found for RMB0.55 ($0.09). Quite cheap, but the datasheet's only available in Mandarin. I used a translator app to OCR it.

I managed to lower the range down to 2.0m +- 0.2m by modifying the 1Mohm across the Opamp to 145kOhm. I don't have any idea how the relationship of range and resistor value scales in this circuit so I just tried a bunch of different values until I got it right. I think it works similarly to the RCWL-0516's R-GN resistor. 

Anyway, just thought you guys might be interested in my little learning experience.

lluotx, 2023-03-05:
I am nterested in my little learning experience and want to increase the ON delay time.  Could you send me the HYL-1617 Mandarin datashheet? I can't find it on any website. Thank you.

tiltec, 2024-01-05:
I also tried to increase the ON delay time - from reading the manual on Taobao (via google translate), I understood that increasing the 680k resistor between VDD and CT should increase the ON delay time. I put 1M there, but it didn't turn on afterwards. I guess there's some limit how much it can handle? Or I made an error while soldering.

From the manual I also got that after every ON period, there's a mandatory OFF period that always is 0.07x as long as the ON period.

Anyway, maybe the manual on Taobao is also wrong, because it mentions MA5002 all over the place.

cvdabbeele, 2024-02-03:
@Halo3fanz thank you for your amazing work on drawing the schematic  
@tiltec I also wanted to extend the duration time that the lamp remained on.  
I was more lucky, it worked for me
R=1M,C=1nF->t=40Sec
R=1M,C=3nF->t=120Sec 
R=1M,C=10nF->t=6,5min=400Sec  (but the hold-off time became too long)

Another drawback of this implementation is that it is not re-trigger-able.  Once triggered, it will go off after the configured time, regardless if it detected movement during the "on" time.
I have not looked into that, but I want to fix it... it is annoying 

 [image]

[image]

/chris

cvdabbeele, 2024-02-03:
Update: 
Disconnecting pin4 from the pcb and connecting it via 10K ohm to pin 8 makes it retriggerable
I have the impression that this works so much better... no need to change the RC timing.
I see that pin 4 is also connected to the LDR, so I'm not sure if this modification is going to mess with the light sensitivity
I'll do some filed testing later
[image]
/chriV

cvdabbeele, 2024-02-27:
Update:
the modified lamp has now been working for a couple of weeks.
It is in a dark room and has to light up multiple times per day.
It nicely re-triggers, as long as it sees movement, and the light has regularly been "on" continuously for 15 min without me having to wave my arms or anything.  
So far, no negative effects on this hack
/chriV

qqyy30, 2024-11-24:
Thanks, one comment, to make it re-triggerable, just need to move below RES  from left to the right.

> Update: the modified lamp has now been working for a couple of weeks. It is in a dark room and has to light up multiple times per day. It nicely re-triggers, as long as it sees movement, and the [...]
[image]