jdesbonnet/RCWL-0516 #57: Low sensitivity

ShayGus, 2022-04-21:
I've read many articles about this thing being tooo sensitive.
I've rigged it with an esp8266 chip, from my observation, if way less sensitive the I expected it to be.
Is it normal for this board to not react to small movements such as turning of the head or raising of the arm from an arms length distance?
I was actually looking for a device that would react to such movements.

dpordomingo, 2022-07-12:
Is your device detecting any movement further than 3cm? I tried with many Arduino and ESP8266, and it only detects movements when they are pretty close (~3cm)
(With no resistors nor capacitors added in its back side; it's just as it came from Aliexpress)

gonzacarv, 2022-08-08:
Same issue here. I have 5 of them, they react only to movement within 3 - 5 cm. No resistor added.

Have you found a solution?

LorenKonkus, 2022-08-23:
Yup, I've found that these devices are very sensitive to surroundings.

- On my bench with an ESD mat, they only detect movement within a couple inches or so. 
- On a protoboard, they only detect within an inch or so.
- When in the same room as an older wifi router, they only detect within a foot or so.

Isolated from all that, with 12 inch leads so it's not next to an arduino board or power supply, and it'll detect a small head nod from across a room.

Wesker94, 2022-12-11:
> I managed to detect movements till 5m far away by connecting it by wires to the Arduino, instead of attaching it directly to the protoboard. I guess that when it's directly attached to the protoboard, it kind of interferes its normal behavior

Yes! It is true, in fact, in the RCWL-0516 (GIT) information, it mentions this point, "the sensor must not be obstructed the front side with anything metallic. The rear side must have a clearance of more than 1 cm from any metal" and The breadboard is made up of metallic lines that are less than 1 cm apart when you put the sensor on it. So this create an issue behavior

To avoid this, connect the sensor via cables and it has sensed up to 4.96 meters.

This is my experience, I hope will help you

mibcat, 2022-12-23:
Same issue for me. Bought 4 devices which all shows only a few centimeter detection range.
Followed all the tips - but none is working
- connect the device via cable to have a distance to any metal objects
- power the device on VIN pin in a range from 5 to 25 V
- power the device on 3V3 pin with 3.3V

I found a comment about a different resistor value between a working and non-working device:
https://wolles-elektronikkiste.de/rcwl-0516-mikrowellenradar-bewegungssensor#comment-15511

As stated the working device has a value of 100 Ohm whereas the non-working device has a value of 1k Ohm.

But from the [schematic](https://github.com/jdesbonnet/RCWL-0516#schematic) this resistor R1 is "only" a series resistor in the VOUT line. Hard to believe that this has an influence to the detection range.

However, can somebody confirm this topic ?

goingalong, 2023-03-03:
Like many I found this device lacks sensitivity. Works to about 3cm only so would like to understand how anyone detecting at anywhere near the claimed range is achieving it?

I tested 2 modules both powered from a 9v battery on 30cm leads away from a breadboard. Detector clamped on plastic support, outdoors on 30cm leads. No mcu.

Was hoping to use as part of an outdoors security camera. Has anyone achieved anything similar?

tr1oii2l, 2023-03-09:
> Same issue for me. Bought 4 devices which all shows only a few centimeter detection range. Followed all the tips - but none is working
>
> * connect the device via cable to have a distance to any metal [...]

I also have issues with the very short detection range (< 5cm). Did all the tests similar to yours and still no vail. I have two boards but both not working as intended, both have the 102 SMD resistor @ R1. So if we try to replace the R1 with 100 Ohm, will it work again? (I doubt it because as you suggest, it's just a resistor in series with the VOUT line)

goingalong, 2023-03-09:
> I've read many articles about this thing being tooo sensitive. I've rigged it with an esp8266 chip, from my observation, if way less sensitive the I expected it to be. Is it normal for this board to [...]

I know this question is a bit old now but if you need a module that is sensitive to small movement you might try this: 
https://thepihut.com/products/5-8ghz-microwave-radar-module

I have one that I have yet to try. Be warned though the documentation is very sparse and amounts to code for an Arduino. I planned to use it on a Pi Pico so I have some conversion work to do.

mibcat, 2023-03-10:
In the meantime I also switched to an other sensor: HLK LD2410

50mil spacing connector:
https://de.aliexpress.com/item/1005004848239174.html?spm=a2g0o.order_list.order_list_main.21.21ef5c5fW49x6g&gatewayAdapt=glo2deu

100mil spacing connector:
https://de.aliexpress.com/item/1005005242026136.html?spm=a2g0o.order_list.order_list_main.10.21ef5c5fW49x6g&gatewayAdapt=glo2deu

Is supports 8 distance gates (each of 0.75m) and can detect moving and standstill object (humans :smile:)

casesolved-co-uk, 2023-10-26:
> Yes! It is true, in fact, in the RCWL-0516 (GIT) information, it mentions this point, "the sensor must not be obstructed the front side with anything metallic. The rear side must have a clearance of [...]

In case anyone is confused about which side is front/back:

> The forward side of the board is the side with components. This side should face the objects being detected. Do not obstruct forward side with anything metalic. The back side should have clearance of more than 1cm from any metal.

eyetengu, 2024-06-26:
I am getting pretty reliable results at ~15-16 ft.
Using a super-basic perfboard, the RCWL-0516, red led, (2) 220 resistors(series), barrel jack(female),  and a 9V battery.
The board is mounted perpendicular to the perfboard.
Sensor connections used(GND, OUT, VIN)
- OUT connects to the led and resistors and returns to the barrel jack(-)
- GND connects to barrel jack(-)
- VIN connects to barrel jack(+)

Hook up 9V battery w/ barrel jack attachment.
Allow a minute or so to calibrate.

My results
Line Of Sight
-at about 15-16 ft away the led will come on.
-it seems to register continuous movement. If you keep moving, the led stays lit.
-the led goes after a delay of about 3sec.

Device Hidden Within A Container
-circuit and V source placed in a black plastic food container. clear lid placed on top. 
-container placed at ~5 ft above ground
-movement detected at about 18-20 ft away.

These results have been fairly consistent.

My product was purchased through Amazon