jdesbonnet/RCWL-0516 #59: RCWL-0516 Bouncing

jkknighton, 2022-06-08:
Newbie here.  I'm using an RCWL-0516 with a D1 Mini through Home Assistant to trigger a light on when someone enters a room.  The issue I am having it just keeps bouncing from Detected to Clear every few seconds.  This is the 2nd one I've tried with the same issue, the first one was with an ESP32.  I know I must be doing something wrong or maybe there's something I need to add?  Here is the ESPHome code that I uploaded to the D1 Mini.  Thank you.

```
esphome:
  name: mw_sensor01
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "XXXX"
  password: "XXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Mw Sensor01 Fallback Hotspot"
    password: "kta0sKXAWsgS"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: gpio
    pin: D4
    name: "Kitchen Sensor"
    device_class: motion
    #filters:
      #- delayed_on_off: 4000ms
```

rugene76, 2022-12-10:
that behavior seems to be typical. To handle it, I use node-red for automations and use the "trigger" node. I set the node to retrigger on every detection and send a turn-on message. If no detection is seen over say, a 2 minute period, the node sends a turn-off message to the targeted light. I'm sure the same can be done using the home assistant automation capabilities.

See my recent post ( Unit triggers ok but then seems to saturate and requires a ~60 second cool down time #63) regarding an issue I'm seeing regarding the loss of any triggering at all.

good luck

casesolved-co-uk, 2023-10-26:
Having had a play with some of these boards for a day, it could be caused by AC ripple on your voltage supply to the board. I am powering one off a rectified 240V AC supply via a few resistors and a zener diode (replacing a dead PIR on an outdoor light).

It kept retriggering itself until I put an extra decoupling capacitor across the supply. I used a ceramic 1uF, but more is better. If you have problems and the space put both an electrolytic (10uF?) and ceramic.

The capacitor reduced the AC ripple from >4 volts to ~1.5 volts as measured on a multimeter.