jdesbonnet/RCWL-0516 #63: Unit triggers ok but then seems to saturate and requires a ~60 second cool down time

rugene76, 2022-12-10:
I've got the rcwl-0516 connected to an esp32/esphome and physically dangling in the air. I'm testing it to see how reliably it detects movement. 
Initially it triggers well and within about a 10 ft radius. But after about 5 minutes of frequent triggers, it loses all sensitivity and seems to need about a 60 second cool down before it detects again. (times are approximate)

Feels like perhaps the 0516 difference detection gets saturated or reaches a new equilibrium that fails to trigger. I'm guessing that the ~60 second cool down allows for a decay of the difference signal - but who knows....

Anybody seeing this behavior? or can explain what's happening with a fix? seems like a deal breaker for me for controlling lights.
thanks!
rugene

benjaminBoehme, 2023-02-13:
I starting to see a similar behaviour. It's hooked up to a rasberry and I wrote a simple python script which confirmed my observations. Imagine me standing 2m right in front of my sensor an continuously waving my arm. What I'm seeing on the output is:

```
┌──────────┬──────────────┬──────────────────────────────────────────────────────────────────────────────────────────────────┐
│          │              │                                                                                                  │
│          │ yes          │       ┌───────────────────────────────────────────────────────────────────────────────────────   │
│ Movement │              │       │                                                                                          │
│          │ no           │  ─────┘                                                                                          │
│          │              │                                                                                                  │
├──────────┼──────────────┼──────────────────────────────────────────────────────────────────────────────────────────────────┤
│          │              │                                                                                                  │
│          │ activity     │       ┌─────────────────┐      ┌────────┐     ┌───┐                           ┌───────────────   │
│   OUTPUT │              │       │                 │      │        │     │   │                           │                  │
│          │ no activity  │  ─────┘                 └──────┘        └─────┘   └─────────  ──  ──  ──  ────┘                  │
│          │              │       ▲                 ▲      ▲        ▲     ▲   |                           ▲                  │
├──────────┴──────────────┼────── │ ─────────────── │ ──── │ ────── │ ─── │  (15.2) ───────────────────── │ ─────────────────┤
│                         │       │                 │      │        │     │                               │                  │
│   Time (seconds)        │      (0)               (6)   (8.4)   (11.2) (13.5)                          (90)                 │
│                         │                                                                                                  │
└─────────────────────────┴──────────────────────────────────────────────────────────────────────────────────────────────────┘
```
In other words: although there's constant movement i can see a decline in the maximum pulsewidth on the output which finally falls slightly below the calculated value of `Tx` which is around 2.4 seconds (see [this picture within this repository for calulation and definitions](https://github.com/jdesbonnet/RCWL-0516/blob/master/doc/john_taylor_rcwl-0516-1.png)). After that the sensor becomes "blind" for a slightly variable amount of time (somewhere between 30s and one minute) and returns back with reduced or full sensitivity* (did no find a clear timing pattern here). At the same time the pulsewidths on the output are declining, I can observe a slight decline in `Ti`which should be around 2.4 s as well.

*I do not observe a reduced sensitivity in terms of range, but in therms of the maximum duration of the activitiy "pulse" on the output. The maximum "initial" pulsewidth I get is around 6-8 seconds with continuous movements.

To me this looks like there's some issue with at least one capacitance not (un-)charging sufficiently within the timeframe it is supposed to. Measurements of the SMD parts do not indicate a fault to me - at least their values align with the datasheet's values besides R1 which is 1kOhms on my board. So I guess it must be some issue within the IC.

rugene76, 2023-02-13:
nice investigating. I guess the part is "too good to be true" for its price point.

I did find that conditioning the 5v supply can greatly reduce false triggers. I ended up putting a 10uf electrolytic followed by a ferrite bead and then a 100pf ceramic capacitor on the 5v feed to create a low pass filter over a wide bandwidth. Even though that helped a lot, I'm still getting weird behavior similar to what you describe. Unfortunately, I don't have an oscilloscope capable of delving into this more deeply.
So after much experimentation, I've given up on the part for use as a presence detector in an ESPHome environment.

casesolved-co-uk, 2023-10-26:
I wonder if the chip has some kind of automatic gain control which isn't quite tuned correctly.

I bought 3 of these boards which were still panellised so must be identical: one worked with several meters of range straight away, one worked with about 10cm of range and the other seemed to have a pretty good range too the second time I tried it.

BUT if you leave it running the range seems to vary a lot. Sometimes it works great, other times you have to get within cms.

I also wonder if there's actually a fault with the design which is why they are so cheap? Shame.

casesolved-co-uk, 2023-10-26:
Also see https://github.com/jdesbonnet/RCWL-0516/issues/57
Could be some kind of periodic interference with metallic objects close-by?